Ultimaker/Cura #4733: Skipped layers in a somewhat complex model

mclroy, 2018-11-05:
**Application Version**
3.2.1

**Platform**
Fedora 27

**Printer**
Prusa i3 Mk2

**Steps to Reproduce**
Just take for example https://www.thingiverse.com/thing:125344 and see multiple skipped layers.

**Actual Results**
A lot of skipped layers and tons of filament lost in action. :(

**Expected results**
Good print according to a model.

**Additional Information**
[image]

Infill in place of skipped layers:
[image]

Skipped layers:
[image]

G-code loaded and the skipped layers are visible:
[image]

smartavionics, 2018-11-05:
And Aldo also says...
[image]

mclroy, 2018-11-05:
I doubt a general Cura user understands what "watertight" means and how to fix it. Frankly, I don't know.
If a model is not suitable for 3d printing then Cura should tell it in the big red blinking letters otherwise the whole situation is misleading.

X-Ray view:
[image]
[image]
No parts in red?

With "Union Overlapping Volumes" turned off:
[image]

In Solid view it looks just awesome:
[image]

BTW the rocket fins are not printed at all:
[image]