Ultimaker/Cura #4734: Cura 3.5.1 Only prints Skirt/Brim and support structure, no model

BristolFD152, 2018-11-05:
**Application Version**
3.5.1
**Platform**
Win10

**Printer**
Tevo Tarantula

**Steps to Reproduce**
Printing any model

**Actual Results**
Only the adhesion layer prints and the support structure.  The model does not print at all  When I print any of my old/saved print projects, those work perfectly.

**Expected results**
An actual model should be printed

**Additional Information**
A couple of upgrades ago, I started to get errors that my profiles contained bad settings and that I should reset all the profiles.  I decided to reset them for 3.5.1 to start over fresh.  Now I cannot print any models.  I can only print support structure and skirts/brims.

It is not a clog as the skirts and brims print correctly.  Saved projects print just fine.  Anything created with 3.5.1 will not print.  I can't figure out what setting I'm missing.  During the prints from 3.5.1, the extrusion gear won't even move to advance the filament.

[TT_X.zip](https://github.com/Ultimaker/Cura/files/2550570/TT_X.zip)

nallath, 2018-11-06:
Could you try the 3.6 beta? We fixed some related issues there.

pynej, 2018-11-06:
Possibly a issues with the multi-extruder settings? Maybe it thinks theres a second extruder for the model material.

nallath, 2018-11-06:
I looked at his project file, there is no extra extruder active, so that's probably not it.

Ghostkeeper, 2019-02-19:
Can't reproduce this here either.

BristolFD152, 2019-02-20:
I've isolated the issue to temperature setting of the print.  When I lower the temp below 170, the issue occurs.  Something happened between now and when I first bought the printer.  I was able to print excellent prints at 165 for PLA.  Now, when I set to 165, I get the issue where I cannot print the model, only the raft and support structure.  I set to 170 and the print works, but the model isn't great quality, you can tell the print temperature is too high.  The filament is a little too runny and doesn't look great.

Ghostkeeper, 2019-02-21:
Wow, 165 is really cold for PLA I'd say! It's usually more like 190.

So you could see your print in layer view, but the printer is not actually printing it?
Maybe the difference is in the print speed then. If it's printing faster, then it does extrude?

fieldOfView, 2019-02-22:
I think the printer firmware has a cold extrusion protection that kicks in below 170 degrees. The bed adhesion is printed, because  by default this is printed 5 degrees hotter than the rest of the print.

BristolFD152, 2019-02-22:
The temperature is relative.  I think the printer temperature is off by 15c.  Where I am setting at 160-165 it's actually 175-180c.  The problem is, I need to be able to lower the temp back to 165 again.  I must have changed a setting on the printer that is now preventing it from printing.  When I run prints at 175, it's probably 190c and it's too high for the PLA.  It prints wood filament beautifully.  I set it at 180 (probably 195 actual) and there are no issues at all.

I've had friends create STLs for me to run and I get the same issues.  I've tried 5 different slicers, it has to be the printer that is overriding the filament feed.
  

When I first received the printer, I ran a bunch of test prints.  Setting at 180 was far too high and I ran into all the expected issues of printing too high.  I was able to run prints down to 160 and I found 165 is best for PLA, 180 for wood PLA.

BristolFD152, 2019-02-22:
I did lower the print speed to 50, I'll try bumping it back to 60 to see if that affects it.  I remember testing print speeds to try to get a nice model.  Hopefully that is the issue!

csn495, 2019-05-30:
I began having this problem as soon as I updated to version 4.1.0. I tried multiple different files but all had the same issue. I do not have this issue on 4.0.0.

BristolFD152, 2019-05-30:
That is awesome news! Thank you. I'll try the older version.

closed 2020-06-20 by no-response[bot]