Ultimaker/Cura #4737: Migration from 3.4 to 3.6

Michaels-3dPrinting, 2018-11-06:
**Application Version**
3.6.0 beta

**Platform**Windows 8

**Printer**
Anet A3 or AM8

**Steps to Reproduce**
 Migrate from 3.4 to 3.6 says configuration files have many errors. The main theme seems to be multiple extruders which I have not configured as I only have one.

**Actual Results**
Big list of errors on extruder definitions

**Expected results**
 A simple migration using previous configuration file without introducing extraneous errors

**Additional Information**

Michaels-3dPrinting, 2018-11-06:
Now using firefox explorer does not work with you website
[cura.log](https://github.com/Ultimaker/Cura/files/2552976/cura.log)
At least if did not crash unlike 3.5 - now put in new configuration details for printers

Michaels-3dPrinting, 2018-11-12:
> The fact that you have many extruders is expected, since when you create a "custom fff printer" we create it with 8 extruders because it's the maximum that you can have, even if you only use one.
>
> It shouldn't be any configuration errors from 3.4 to 3.6, but it's not clear what happen without having your configuration folder. Could you please attach also your 3.4 folder (a zip file may work)
>
> Thanks. See if we can find the issue.

Tried to send folder zipped but bounced and rejected as to large  - so now  just supplying config

 

Hope that helps

 

Please advise if you would like anything else

 

Regards

 

Michael

fieldOfView, 2018-11-12:
Sending attachments in emails to github does not work. You have to use the web interface of github.

diegopradogesto, 2018-11-19:
@Michaels-3dPrinting any updates? Are you able to send us the 3.4 folder using the web interface?

AnakinSpain, 2018-11-22:
same problem using OSX.
[cura.log](https://github.com/Ultimaker/Cura/files/2609128/cura.log)

diegopradogesto, 2018-12-10:
No @AnakinSpain, looking at your logs, they look pretty similar at these ones: https://github.com/Ultimaker/Cura/issues/4870, and this issue is fixed in the next release that will come soon!

I'm closing this issue due to inactivity for 3 weeks.

closed 2018-12-10 by diegopradogesto