Ultimaker/Cura #4744: Adjust Retraction Extra Prime for Travel Distance

pynej, 2018-11-06:
Tangentially related to https://github.com/Ultimaker/Cura/issues/3351

Even when printing with rather forgiving PLA there are frequently under-extrusion issues on a larger part after a long move. Increasing the Retraction Extra Prime helps, but if the part also has many small features that cause multiple retractions in a small area then there ends up being a buildup of treasure and skipping in the extruder. 

It seams obvious to me that the Retraction Extra Prime about should be modified by the travel distance. 

In short, a calculation that uses the travel distance as a inverse adjustment to the extra prime amount is what is needed. Possibly implemented with a Retraction Extra Prime Min Travel Distance and Retraction Extra Prime Max Travel Distance that are used to do the calculation, or some other method.

**Application Version**
N/A

**Platform**
N/A

**Printer**
MDF

**Steps to Reproduce**
Print a large model with fine details needing retractions.
~.25mm3 Retraction Extra Prime

**Actual Results**
Details are nice, but under extrusion on the start of the perimeter inner wall.

**Expected results**
A nice model.

Ghostkeeper, 2019-02-19:
See also https://github.com/Ultimaker/Cura/issues/3452. Not the same feature request, but it fixes the same problem.

Ghostkeeper, 2019-02-19:
I've discussed this with a colleague and we decided that it is probably too complex for a user to understand such a setting. It's hard for a user to input a setting value like "0.012" and then understand that it's going to be priming 0.012mm extra filament for every mm of travel. We're still looking for a way to do this (perhaps not configurable) but we're leaning more towards an implementation like in #3452.

closed 2019-02-19 by Ghostkeeper

Thisismydigitalself, 2019-05-03:
https://github.com/Ultimaker/Cura/issues/3452

loneFunction, 2020-02-04:
There was a plugin that allowed for a scalable retraction prime which would address this. It does not work with Cura 4+ for some reason.
https://github.com/Pheneeny/CuraPlugin-ScalableExtraPrime

pynej, 2020-02-04:
The majority of my issues were fixed by using significantly higher retraction distance.

loneFunction, 2020-02-04:
I've been using high retraction distances in order to eliminate stringing, but it has also caused other issues. It seems large retracts with my current hotend design form air pockets inside the melt zone  and then pop out. This causes a lof of surface quality issues as there are pits all over the place. At first I though it was wet filament, but it has been happening with every spool I've tried.