Ultimaker/Cura #4745: Request:  Add all cura settings to the gcode file

FormerLurker, 2018-11-06:
**Application Version**
3.5.1

**Platform**
Windows 10

**Printer**
Prusa Mk2

**Steps to Reproduce**
Slice any STL and save the gcode.

**Actual Results**
No settings appeared within the sliced gcode like Slic3r or Simpli3d does (an older version of cura I had added SOME of the settings to the bottom of the gcode file)

**Expected results**
All settings should appear somewhere within the gcode file.

**Additional Information**
I've been developing a plugin for OctoPrint called Octolapse, and have been enhancing it to work better with specific slicers.  I am able to extract all of the necessary settings Octolapse needs to function (print speeds, retraction distance, zhop height, build volume, etc) from both Simplify 3d and Slic3r, but I do not see the relevant settings within gcode produced by Cura.

Adding all settings to the gcode file also makes it easier for people to get assistance online, since posting the gcode file itself is enough to illustrate all relevant settings needed to diagnose problems.

The older version of cura I had DID include some settings towards the bottom of the gcode, but they are in a format that is not human readable.  While working on a parser to digest the settings, I realized that not all of the relevant information was included in the settings anyway, so I stopped support for settings extraction from Cura, which is really a shame.

If you feel that the settings clutter the gcode, perhaps consider adding an option to cura to output the settings, preferably in a human readable format (one setting per line, embedded within a comment).  However, I'm guessing that most users will not turn this option on during their first use, which is the most likely time they will be asking for assistance and having other general 3d printer problems that are common among new users.  If you add an option to output the settings, I strongly feel it should be enabled by default.  Just visit the 3d printing sub on Reddit.com to see how difficult it is to extract the information necessary to assist a user with their issues.

Thank you for listening, and thanks for your work on Cura!  It's an excellent slicer.

diegopradogesto, 2018-11-08:
We dump the settings that are in a custom profile or the settings that has changed (what we call user changes). If you are using an existing profile, you don't get this (as far as I remember).

Please provide a project file so we can reproduce the issue with your exact environment and it would be helpful if we need to fix something.

Chalmes, 2018-11-08:
The idea here is that the gcode should contain ALL the settings, irrespective of whether they are changes from a cura profile or not so that external processing can read those settings to make various decisions.

While adding settings that are differences from a current profile might work for Cura, it won't work for anything else.

FormerLurker, 2018-11-08:
@diegopradogesto, @Chalmes has the right idea here.  What I am requesting is that ALL cura settings are added to the gcode file somewhere, just like Slic3r and Simplify3d do with their own settings.  An alternative would be to provide an option to include all settings within the gcode file.  It does not help me to know which settings have changed from the profile defaults, unless ALL of them have changed.  I will not have access to the cura profile, only the gcode.  This is typical of many scenarios, and not just my use case.  I'm happy to elaborate if that would help.

If you need any additional information, I would be happy to provide it.  I used the default 'Prusa I3 Mk2' profile that came with the 3.5.1 version with no modifications to any settings.  I sliced a simple test cube and inspected the gcode for settings, but none were contained within the gcode file.  I expected ALL settings to be there, but none were.  It is quite possible that 'user changes' would appear if I changed the defaults, but that is not what I am requesting.  If it would be useful to you I'd be happy to change some settings to verify that they are added to the gcode file in some format.

Thanks again for reading over my submission.  I know it is time consuming, and I appreciate it a lot.

FormerLurker, 2018-11-13:
I believe this is the information you requested, but please let me know if you need anything else.

This is the printer profile I'm using (default MK2 profile with no modifications):
[prusa_i3_mk2.def.zip](https://github.com/Ultimaker/Cura/files/2576930/prusa_i3_mk2.def.zip)

Here is a sliced gcode file which contains no settings:
[PI3MK2_20mm-box.zip](https://github.com/Ultimaker/Cura/files/2576924/PI3MK2_20mm-box.zip)

diegopradogesto, 2018-11-20:
I understand the point. We had something like this before, but we remove it for several reasons, including the increase of saving time and increasing of gcode size. I discussed it with the rest of the team and we decided not to do it.

However, we thought that it would be a nice proposal to have this as a postprocessing script, something that you can enable or disable when saving your gcode.

FormerLurker, 2018-11-26:
@diegopradogesto,

Thanks for reviewing the request, I appreciate it!  Should I open a new issue regarding a post-processing script, or do you envision this being a third party thing?

In the meanwhile I have implemented a workaround for my plugin, but it means that people using Slic3r and Simplify3D will have a plug-and-play experience, but users of Cura will need to manually configure the plugin both at setup and after any Cura settings changes.  This is unfortunate, but it really just means that the UX won't change for Cura users.

I do want to mention a few things regarding your objections, just for posterity and completeness in case you need to refer to this issue in the future.  I just compared file size of a gcode sliced with Simplify3d including settings and without settings.  The file size difference was only 6.85KB.  Assuming every gcode file has the same number of settings, this means that one would need more than 145 gcode files before the settings occupied 1MB of disk space, which seems pretty trivial.

Cheers, and thank you for all of your efforts and for your excellent slicer!

diegopradogesto, 2018-11-27:
At this moment, we are quite busy working in the new redesign and probably we won't have time for implementing the post-processing, so we will wait until any contributor want to help. 

For your plugin, it seems that you only need to dump a few settings, not all of them, so I'm thinking that maybe you could create a PR for only include those in the GCode. Just thinking out loud.

FormerLurker, 2018-11-27:
@diegopradogesto, you are correct, I only need some of the settings.  I'll start looking into extracting those settings.  Thanks again for your thoughts!

tjjfvi, 2019-01-04:
@FormerLurker: If you are still working on this I might recommend using [OctoPrint-SlicerSettingsParser](https://github.com/tjjfvi/OctoPrint-SlicerSettingsParser) as an optional dependency, which adds metadata with the slicer settings. That way you can pull away the parsing logic to another plugin, as it isn’t core functionality of your plugin. While it doesn’t currently support Cura, I am working on it and can let you know once I have finished.

*Disclaimer: I am the creator of that plugin.*

FormerLurker, 2019-01-08:
Hi @tjjfvi, thanks for your post and for your great plugin!  I've gotten a few messages already about your plugin from some other users :)

Believe it or not I created routines to parse the settings for Simplify and Slic3r already, though your method is much simpler.  I'm going to compare speed between the two.  I'll let you know how it turns out!

Also, please create an issue on the Octolapse github pages for this, that way we can communicate without sending update messages to the good people in the cura git :)