Ultimaker/Cura #4746: Problem printing 3 line width cuboid

zz9pa, 2018-11-06:
**Application Version**
3.5.1

**Platform**
Windows

**Printer**
CR10 mini

**Steps to Reproduce**
Generate a STL with a vertical cube 1.2mm wide, print with .4 nozzle and line width, and "lines" for top/bottom pattern

**Actual Results**
Prints outer edge and then uses "lines" to fill the gap, taking ages to print something that should be quick.

**Expected results**
print 3 lines

[image]

Cant upload an STL - but you can generate one using openscad with

cube([1.2,61,39]);

diegopradogesto, 2018-11-10:
Was it different in 3.4 or previous versions? We changed some behaviour in the merge small lines algorithm and I'm wondering if it's is related.

PS. You can upload a STL or project file if you zip the file.

zz9pa, 2018-11-10:
Its fine in 3.4.1
[image]

zz9pa, 2018-11-10:
[test1.zip](https://github.com/Ultimaker/Cura/files/2568489/test1.zip)

zz9pa, 2018-11-13:
Still present in 3.6

closed 2020-06-20 by no-response[bot]