Ultimaker/Cura #4747: Reload all models does not reload 3MF models, and file changes are not detected

rreilink, 2018-11-07:
When placing a model using 3mf file format, and then changing the file using an external program, Cura does not ask to reload the model. After manually selecting 'reload all models', 3mf files are not reloaded either.

With STL model files, this does work correctly.
**Application Version**
3.5.0

**Platform**
Windows

**Printer**
[3mf_reload_bug.zip](https://github.com/Ultimaker/Cura/files/2556480/3mf_reload_bug.zip)

**Steps to Reproduce**
Add a model from a 3mf file. Change the 3mf file using an external program. In Cura select file->reload all models.

**Actual Results**
The model in Cura stays as-is

**Expected results**
Cura should mention that the file is changed externally. After reloading, the model should be updated to the changed 3mf file.

nallath, 2018-11-08:
The problem with this is that a 3mf file can hold multiple models. So what to do when you scale and move one of the models in cura and then make changes in the 3mf?

rreilink, 2018-11-08:
That is a good point. Is there a reference (e.g. name or identifier) in the
3mf format that would allow to read in the new file and match the models in
the updated file to the models in Cura? Or, as a fall-back, map the first
model in the new file to the model in Cura that was the first model in the
original file etc?

As a very minimum, I would expect the option to invalidate (i.e. remove)
models which originate from a modified file and place them again without
transformations, since with the current behaviour, it is quite easy to
print an old version of your model (when the changes are small and you
cannot identify whether the model in Cura is old or new by looking at its
graphical rendering)

Met vriendelijke groet, Kind regards,

*Rob Reilink M.Sc. Ph.D.*
Sr. Mechatronic Engineer

www.ultimaker.com <https://ultimaker.com/en>

nallath, 2018-11-08:
Well, 3mf has ID's, but there is no guarantee that the CAD software actually keeps them the same. 

We've thought about this a while back, but we weren't able to find a good flow that would cover most of the cases, so resolving this was simply postponed in favor of other issues.

diegopradogesto, 2018-11-08:
The reload feature that we use to have in previous versions doesn't exist anymore in Windows, so maybe it has to do with that. Please read my comment in this other thread: https://github.com/Ultimaker/Cura/issues/4633#issuecomment-433025302

diegopradogesto, 2018-11-19:
Hi @rreilink did you check if it works in Mac?

fieldOfView, 2020-05-05:
This has been fixed in #7230

closed 2020-05-07 by Ghostkeeper