Ultimaker/Cura #4752: Suggestion: add check box to prevent retractions while printing infill

kornshadow097, 2018-11-08:
**Application Version**
3.4.1

**Platform**
win 10 64bit

**Printer**
tevo tarantula

**Steps to Reproduce**
 with combing on or off, there are way too many retractions happening. i keep seeing dozens of retractions occur with combing mode on not in skin. some say if its off they get infill retractions but i get them regardless. I suggest that there be a check box that allows us to turn off retractions when moving to print infill lines, while already printing infill lines.

If the line its printing next is not infill, sure it can retract. But as long as the next line to be printed is infill, then it needs to stop retracting so much. 

**Actual Results**
its seriously a huge waste of time. Honestly most of my infill lines get underextruded too because the lines are so small sometimes and retracting just causes my filament to get chewed up and the ooze just doesnt have time to recover before the next line is due.

**Expected results**
 this should improve infill line quality, lower print times by around 20% depending on the project (my current 5hr prints would reduce by 45min-1hr). also the filament would be better off and my extruder titan extruder wouldnt get covered in filament dust every time i print infill...

**Additional Information**

smartavionics, 2018-11-08:
Please zip up a project file and attach to this issue so we can check your settings. You should not be getting retractions while printing infill unless the settings are asking for it. Thanks.

Ghostkeeper, 2019-02-19:
We can't reproduce this. If combing is set to not in skin it makes retractions when moving through skin, but as long as it's moving only through infill it's not retracting in the models I've tried.

closed 2020-06-20 by no-response[bot]