Ultimaker/Cura #4754: suggestion: add x/y seam alignment

kornshadow097, 2018-11-08:
**Application Version**
3.4.1

**Platform**
win 10 

**Printer**
tevo tarantula

**Steps to Reproduce**
instead of just a z seam alignment, there should be x/y alignment as well. z seam alignment helped solve the ugly lines i use to get on z, but i still get them on x/y because things always start and finish on the same area, which causes a  long scar looking artifact. i didnt get them until i downgraded from 3.5 and 3.5.1  due to bugs with the profiles, having to do with nozzle sizes crashing the app. 

**Actual Results**

**Expected results**

**Additional Information**

id make it easier for us to contribute our feedback and suggestions via the app too. i could add so many upgrade ideas if it were as easy as having a button like in mobile apps that allows me to just send a suggestion. vs logging into here and having to fill out a generic form asking me stuff i already answered a dozen  times before..

Ghostkeeper, 2019-02-19:
I don't understand this feature. The slice doesn't always finish in the same part as what it started on. In fact, if you have more than 1 part, it should never finish on the same part that it started on, because it completes one part at a time.

It also doesn't always start in the same part that the previous layer finished on, because that part may not exist on the next layer, or be merged or split.

If I understand your issue well, you want the nozzle to end up at the end of a layer in the same X/Y location as where it needs to start in the next layer?

The scars you're seeing may be solved by setting the combing mode to "not in skin".

GregValiant, 2024-11-24:
@kornshadow097 A lot of work has been done to the seams in the current Cura versions (5.8.0 and up).
I'll go ahead and close this as it isn't likely to be relevant anymore.

closed 2024-11-24 by GregValiant