Ultimaker/Cura #4760: [3.5.0] Gaps fill

Liger0, 2018-11-08:
**Application Version**
3.5.0

**Platform**
Windows 10 x64

**Printer**
Cr-10S

**Steps to Reproduce**
Use any model.

**Actual Results**
 It won't fill most of the big gaps, but it will fill the very small ones

**Expected results**
It fills the small gaps altogether with the small ones

**Additional Information**

[image]

diegopradogesto, 2018-11-10:
There is a setting that will ignore the tiny gaps: "Filter out tiny gaps". However that has nothing to do with bigger gaps. Can you please attach the project file?

Liger0, 2018-11-10:
> 
> 
> There is a setting that will ignore the tiny gaps: "Filter out tiny gaps". However that has nothing to do with bigger gaps. Can you please attach the project file?

[CCR10S_base.zip](https://github.com/Ultimaker/Cura/files/2568498/CCR10S_base.zip)
As far as I see, the gap infill is only done between 2 greeb peimeters or between the yellow perimeter and the green one. It isn't done between the infill and yellow perimeters.

GregValiant, 2024-11-15:
Is this still an issue with current Cura versions (5.8.0 and up).  Can this be closed?

closed 2024-12-09 by github-actions[bot]