Ultimaker/Cura #4762: Material is lost after attempting to set it to an unsupported diameter

mqudsi, 2018-11-09:
Under Cura 3.5.1 with the Ultimaker 3 Extended as the current machine and the Print Core set to AA 0.4 or AA 0.8, attempting to create a new material (by duplicating an existing one) results in broken behavior when changing to a non-standard diameter (such as 1.75mm).

Steps:
1) Configure Cura for Ultimaker 3 Extended and AA 0.4
2) From the materials drop-down select "manage materials"
3) Duplicate the selected material and give it a unique display name so you can monitor what happens
4) Change the diameter from 2.85mm to 1.75mm. 
5) A dialog appears stating that the current extruder is not compatible with the selected diameter, and asks if you want to continue.

Expected behavior:
* By clicking "yes" I force the non-standard diameter.
* Clicking no -> undoes the change I made to the diameter

Actual behavior:
* Clicking "yes" deletes the newly-added material from the list of materials
* Clicking "no" keeps my changes (!!) but then the material disappears from the list of materials

mqudsi, 2018-11-09:
Please re-open this. It is not a duplicate of #4582 and persists in 3.6 beta. 

This has to do with a bug in the manage materials dialog that affects first-party (ultimaker) print printers and their print cores.

mqudsi, 2018-11-09:
Any fdm_material files with a diameter != 2.85 (including the one you are currently editing) immediately disappear from the list and the configuration ends up in a corrupted state with some settings from the previously-chosen-now-disappeared profile and some settings from the now-magically-selected profile.

diegopradogesto, 2018-11-09:
Oh, sorry, I read quickly and immediately thought that it was the same issue. So, it seems that the buttons behave in reverse order (weird). I'll check later and try to reproduce it.

mqudsi, 2018-11-09:
Not just reverse order. The material shouldn’t disappear like that and there’s no way to override the material not supported.

diegopradogesto, 2018-11-09:
The material doesn't disappear, it only shows in printers that have an extruder compatible with this material diameter (approximate diameter). For instance, follow these steps:
1. Add printer that uses 2.85mm (P1)
2. Add printer that uses 1.75mm (P2)
3. With P2 being the active printer, go to material manager (notice that all the materials in the list have 1.75mm)
4. Create material (M2): this material has 1.75mm
5. Change material diameter to 2.85mm
6. A dialog warns you that the material is not compatible with your current printer.
7. Accept the risk, click yes.
8. Oh, no!!! M2 disappear from the list (**this is expected**)
9. Switch to your P1 printer
10. Go to material manager.
11. Yes!!!! Material M2 is in the list (**this is expected**)

If you have a different behaviour, then something is wrong. If not, then is just a misunderstanding.

mqudsi, 2018-11-09:
I only have the Ultimaker.

The problem is number 8. Number 7, which we are in accord with is "Accept the risk" means "accept the risk of using this material with this configuration with this printer". So why would the material disappear?

Additionally, changes you make to the material that disappeared are misapplied to the wrong material. 

(It's also poor UX to disappear the material rather than grey it out or disable it or present a message "this material isn't supported by this printer".)

Long story short: I *know* the 1.75mm material works with my Ultimaker with the 0.8mm nozzle. I know because I've used it in past versions that didn't try to outsmart me. How can I use it?

mqudsi, 2018-11-09:
Also you didn't address the use case for "do you want to continue" -> No which is buggy (but not what I want in all cases).

diegopradogesto, 2018-11-09:
> The problem is number 8. Number 7, which we are in accord with is "Accept the risk" means "accept the risk of using this material with this configuration with this printer". So why would the material disappear?

No, it doesn't mean that, it says that this filament is not compatible with you current printer. It doesn't mean that you can use it even thought your printer uses a different material diameter.

> Long story short: I know the 1.75mm material works with my Ultimaker with the 0.8mm nozzle. I know because I've used it in past versions that didn't try to outsmart me. How can I use it?

Yes, the feeder will move the filament (probably not correctly) and the print core will melt the filament as well (not the right amount). However you have big chances of getting underextrusion. Even more, why would you use 1.75mm in a 2.85mm nozzle?? 😕 

I'll check tomorrow, but if the behaviour is the same as I describe, I don't think we are gonna change that, sorry.

mqudsi, 2018-11-09:
The UX here needs work. “Do you want to continue” should not result in the material vanishing.

mqudsi, 2018-11-10:
> Even more, why would you use 1.75mm in a 2.85mm nozzle?? 

The nozzle isn't 2.85mm, it's the 0.8mm nozzle. But the same happens even with the 0.2mm nozzle. 

(As to why I'm using 1.75mm filament *at all* it's because it's more popular and some of the more esoteric materials only come in 1.75mm)

Azimut41, 2018-11-10:
Faced this problem. The creation of a second printer with a diameter of 2.85 helped. As it is written here https://github.com/Ultimaker/Cura/issues/4762#issuecomment-437517708
Initially, KURA puts the wrong size.

surfrock66, 2018-11-15:
FYI this is also documented in this issue https://github.com/Ultimaker/Cura/issues/4550

closed 2026-09-06 by GregValiant