Ultimaker/Cura #4763: 5 extruders bug

prahjister, 2018-11-09:
**Application Version**
3.4 3.5.1 3.6 beta

**Platform**
 Win 10

**Printer**
 
[10mm.zip](https://github.com/Ultimaker/Cura/files/2564538/10mm.zip)

**Steps to Reproduce**
 5th extruder is ignored doesnt show up in gcode

**Actual Results**
no mention of 5th extruder in gcode

**Expected results**
 print 5 colors. 2 real and 3 virtual

**Additional Information**

[10mm.zip](https://github.com/Ultimaker/Cura/files/2564542/10mm.zip)

prahjister, 2018-11-09:
OK after letting the print go it is only the first tool change that is wrong. Instead of tool 0 printing as first tool it is tool 1...after that it is correct. All subsequent layers are correct.

prahjister, 2018-11-09:
;FLAVOR:Marlin
;TIME:4162
;Filament used: 1.55143m, 1.38977m, 1.30941m, 0.756747m, 0.537915m
;Layer height: 0.2
;Generated with Cura_SteamEngine 3.5.1
T0
M140 S50
M105
M190 S50
M104 S215
M104 T1 S215
M104 T2 S215
M104 T3 S215
M104 T4 S215
M105
M109 S215
M105
M109 T1 S215
M105
M109 T2 S215
M105
M109 T3 S215
M105
M109 T4 S215
M82 ;absolute extrusion mode
M163 S0 P0.75
M163 S1 P0.25
M164 S2
M163 S0 P0.50
M163 S1 P0.50
M164 S3
M163 S0 P0.25
M163 S1 P0.75
M164 S4
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
G92 E0
G1 F200 E3
G92 E0
G92 E0
G1 F3600 E-6
;LAYER_COUNT:50
;LAYER:0
M107
G0 F6000 X71.39 Y100.265 Z0.3
;TYPE:SKIRT
G1 F3600 E0

prahjister, 2018-11-09:
After analyzing the gcode more i think i found the issue

If i add a T0 just after G28, the first tool being used is accurate. Now the question is....What is at fault Marlin or the Cura?

It is obvious that T0 is in the gcode but with all the commands after Marlin is confused.

diegopradogesto, 2018-11-10:
Can you please take a look at your start gcode, in your machine settings? Maybe it's related with that.

Can you also attach a project file with this specific scenario, so that we can reproduce it?

nallath, 2019-03-01:
Closing due to inactivity: Aditional information was requested but not provided

closed 2019-03-01 by nallath