Ultimaker/Cura #4765: Feature Request: Marketplace "Restart" instead or add. to "Quit"

Drayson80, 2018-11-09:
When installing Plugins, its a bit uncomfortable to hit quit, then navigate and restart the application.

Suggestion: modify the "quit" button to automatically restart Cura after Plugin installation. Worst case keep "quit" and add a second "restart" button.

ChrisTerBeke, 2018-11-09:
This is technically very hard because the OS has to restart Cura, not Cura itself, so we have little control over this (unless we write something that is running on the OS in the background that does all this stuff, but that's annoying as it's different per OS and you get permissions issues etc.).

diegopradogesto, 2018-11-10:
@LipuFei and I were playing around with some Windows API calls (for lock files). We thought that maybe Cura could use it for this purpose, but we didn't try yet.

Ghostkeeper, 2019-07-09:
We could have a look at the `execl` and related functions: https://docs.python.org/3/library/os.html#os.execl