Ultimaker/Cura #4767: Project file doesn't respect positioning

BagelOrb, 2018-11-09:
**Application Version**
3.5.1

**Platform**
Linux Mint

**Printer**
UM3

**Steps to Reproduce**
Install Mesh tools from Aldo
Load [hex_tiles.stl.zip](https://github.com/Ultimaker/Cura/files/2565514/hex_tiles.stl.zip)
scale to 10%
Split into parts
( Try disabling all preferences related to scaling and positioning )
save project
clear build plate
load project

**Actual Results**
Objects are all over the place

**Expected results**
Objects retain their positioning

**Additional Information**
Before save:
[image]

After load:
[image]

Project:
[repositioning_problem.3mf.removeextension.zip](https://github.com/Ultimaker/Cura/files/2565520/repositioning_problem.3mf.removeextension.zip)

BagelOrb, 2018-11-09:
Here's my preferences:
[image]

diegopradogesto, 2018-11-11:
It's really weird, but I can confirm it. I also attach another small STL that consist in 4 small parts and something similar happens.
[objects.stl.zip](https://github.com/Ultimaker/Cura/files/2569324/objects.stl.zip)

diegopradogesto, 2019-02-15:
Do you have any ideas about this, @fieldOfView ? We saw that it only happens after using the 'Split object into parts'

fieldOfView, 2019-02-19:
I am not sure, but it could be that all of the created meshes have the same origin; the origin of the original non-split mesh. If the mesh origin is somehow included in the convex hull of the now individual meshes when loading the 3mf file, and this is used to push the models apart even though the preferences say not to do that, then this spreading apart would be an expected outcome. I remember issues about both these phenomena, but don't remember the outcome.

An interesting way to test this hypothesis is to select all subparts after loading them back in and merge them. They should snap back to their correct relative position. If this confirms the hypothesis, then Cura is doing two things wrong: (1) it makes the origin of a mesh in a 3mf part of the convex hull and (2) it rearranges models after loading even if the push apart option is turned off.

diegopradogesto, 2019-02-19:
OK, we will investigate. Thanks for the tips.

Devs. CURA-5931

fieldOfView, 2019-02-19:
For the record, these are the related issues I was thinking about:
https://github.com/Ultimaker/Cura/issues/2132 / https://github.com/Ultimaker/Cura/issues/3847 and https://github.com/Ultimaker/Cura/issues/3201

diegopradogesto, 2019-04-26:
I can confirm that after merging the model, they snap back to the original position.

LipuFei, 2019-08-29:
@diegopradogesto @fieldOfView I made a PR to fix this issue: https://github.com/fieldOfView/Cura-MeshTools/pull/8

LipuFei, 2019-08-30:
I also found that one part in the hex tiles model seems to get split too much.

[image]

BagelOrb, 2019-08-30:
That's just because I modelled it badly. Not water-tight.

diegopradogesto, 2019-09-05:
We were taking a look at this issue and after some trials and errors, we've decided to defer this for now at least.

BagelOrb, 2019-09-05:
From my point of view this has nothing to do with the split into parts plugin.

If I save a build plate I expect it to be loaded the same as I saved it, but somehow there is some discrepancy and the model doesn't get loaded the same way it is saved - or not all data used to determine the positions of objects is saved to the project file.

mahtDFR, 2020-01-22:
Not a fix, but a workaround: close file, open Cura preferences and uncheck 'automatically drop models to build plate'. Then reopen the project file.

BagelOrb, 2020-01-22:
Wait a minute. If it is still an issue and it clearly is a bug, then shouldn't it be deferred, rather than Wont fix?

mahtDFR, 2020-01-23:
Ok, I will switch it back to deferred. It's unlikely we will be able to devote time to fixing it.

closed 2023-06-13 by github-actions[bot]