Ultimaker/Cura #4768: 3.5.1 Printspeed for smal Objects

lwbsbde, 2018-11-09:
Some Models also consist party of smaller structures. How can i set this smaller structures to print slower and maybe more Fan speed? for a better quality

diegopradogesto, 2018-11-10:
Take a look at the settings in "Cooling" category. For instance "Minimum Layer Time" and "Minimum Speed" will help you on printing small parts.

I also suggest you to download the "Cura Settings Guide" plugin in the Toolbox. It is a plugin that explains many of the settings within Cura, with descriptions, pictures and animations. It's really helpful.

lwbsbde, 2019-01-12:
But  the Minimum Layer Time is aimed at the complete Layer. If I have in one Layer small parts and big parts, I would like to set a minimum time for every wall . For example 0,5 second.

lwbsbde, 2019-11-22:
Nach einem Jahr steht die Funktion endlich zur Verfügung. 

Danke!

Leider fehlt noch die anpassbare Lüftergeschwindigkeit.

closed 2020-11-03 by nallath

Ghostkeeper, 2020-11-05:
(See the Small Feature Speed setting.)