Ultimaker/Cura #4772: Cura 3.5.1 Save/Prepare(unchecked auto slice automatically) Button is No Response!

yangqxliuy, 2018-11-09:
I downloaded Cura 3.51  in Windows 7 and unchecked auto slice automatically. When I clicked "Prepare" button, there is no reponse. Then I checked auto slice automatically, "Save" button is no response. Why and how to correct it. Please email me the solution. 

Thanks.

diegopradogesto, 2018-11-10:
We have heard before that sometimes is a bit laggy and it takes a bit to change the status. Please click Prepare and wait a bit without clicking again. If still nothing happens, please send us the log files.

Chalmes, 2018-11-12:
Can confirm this is laggy if an object has been sliced prior.  The button doesn't update, BUT it is being sliced in the background, and eventually the UI catches up.

diegopradogesto, 2018-11-19:
Yes, we have some laggyness in some computers, it's hard to test here. @yangqxliuy is it the same problem for you?

closed 2020-06-20 by no-response[bot]