Ultimaker/Cura #4774: Feature request: Remeber last open path

rajinder-yadav, 2018-11-10:
It would be nice to remember the last folder a file is loaded from. Right now, I need to navigate several folder to get to my STL files.

When I close Cura down, I would expect it remember the last folder open so I can load other STL files.

diegopradogesto, 2018-11-10:
Which operating system are you using?

Thisismydigitalself, 2018-11-11:
Same wish and if I'm allowed to add: Saving project should allow me to include the relevant stl's as well.

diegopradogesto, 2018-11-11:
I've just tested on both Mac and Windows, in Cura 3.6 BETA and it works properly. Probably works on Linux as well. @alekseisasin , can you please confirm?

Thisismydigitalself, 2018-11-12:
Ineeded, 3.6B does rememeber last open folder. for me it will make more sense to remember a predefined path/folder, e.g. in my case: 'Projects' where i keep all my work, and such.

ChrisTerBeke, 2018-11-12:
I think remembering the last opened path is OS default behaviour. On my Mac it happens everywhere.

diegopradogesto, 2018-11-19:
@alekseisasin , @Ghostkeeper , @smartavionics can you quickly test in Linux if this is correct? Thanks

zerspaner-gerd, 2018-11-19:
It would not be bad if different storage locations are stored.

I have e.g. a location to store gcode files, a location to open and save Cura profiles, and a location to open stl files.

An example of the program that handles this is S3D.

rajinder-yadav, 2018-11-21:
I can tell you on Linux and in general the default location is set from the application setting like this.

[image]

Ghostkeeper, 2019-07-19:
The save path is remembered, but the open path is not right now.

dstulken, 2019-07-19:
> for me it will make more sense to remember a predefined path/folder, e.g. in my case: 'Projects' where i keep all my work, and such.

I'll second this suggestion!

The existing "Open Recent" action already provides the quick access to the files we are currently working on.  But when moving from one project to the next, opening to our default project folder, rather than the last project's subfolder, would be a huge time saver.

It's also very deterministic in its behavior, which is great for both software devs and users alike.

Ghostkeeper, 2019-07-22:
But then you'd need to have some way to configure the default folder, which is not easy to find/understand UX-wise.

dstulken, 2019-07-22:
How about just a one-line entry in the preferences dialog ('General' section)...  
"Default Project Folder" (or something to that effect).  

If you specify a default using that setting, it uses that path any time you do a File->Open operation.  If you leave it blank, it remembers the prior path you opened.   The default would be blank.

(And if you'd rather not save the prior open path, you could just as easily make that default to the user's home directory instead if that line was left blank.)

dstulken, 2019-07-22:
If it helps, here's how another software package (RawTherapee) handled this situation.
("Image Directory at Startup")

[image]

rajinder-yadav, 2020-06-20:
I guess these was never assigned and left to die?

nallath, 2020-06-20:
It still had a Status: Needs info label. We recently added a bot that closes issues that have that label for at least 14 days. 

In the future, the bot will also remove the label after we added it and the original author adds another comment.

GregValiant, 2026-09-06:
I'm cleaning up these old reports.  I'll go ahead and close this.  If the problem has persisted into 5.13.0 please make out a new report.
5.14.0 will include the ability to hard code a default "Open Project file from location" and a "Save Gcode file location".

closed 2026-09-06 by GregValiant