Ultimaker/Cura #4775: [3.5] Brim inside the model while set to external only

Liger0, 2018-11-10:
**Application Version**
3.5.0

**Platform**
Windows 10 x64

**Printer**
Cr-10s

**Steps to Reproduce**
Use the benchy file.

**Actual Results**
Inside the D letter there is the brim.

**Expected results**
Only the external brim should exist

**Additional Information**
[image]
[CCR10S_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/2568523/CCR10S_3DBenchy.zip)

smartavionics, 2018-11-10:
Duplicate of https://github.com/Ultimaker/Cura/issues/3482.

diegopradogesto, 2018-11-11:
Also duplicate of #4222. That one is still open, so you can keep tracking of that issue.

closed 2018-11-11 by diegopradogesto