Ultimaker/Cura #4776: [3.2.0 - 3.3.1] Remove of software in silent mode prompts a Question

damon-atkins, 2018-11-11:
**Application Version**
3.X most likely all

**Platform**
Windows

**Steps to Reproduce**
Uninstall the software

**Actual Results**
Prompt for "Do you want to remove all your configuration files?" during silent uninstall?

**Expected results**

**Additional Information**
The following needs changing so it does not prompt on silent remove i.e leaves the configs behind.
```
Section "Uninstall"
  Var /GLOBAL CuraVersionDir
  StrCpy $CuraVersionDir "@CPACK_PACKAGE_VERSION_MAJOR@.@CPACK_PACKAGE_VERSION_MINOR@"
  MessageBox MB_YESNO "Do you want to remove all your configuration files?" IDYES remove_user_dir_yes IDNO remove_user_dir_no
  remove_user_dir_yes:
    RMDir /r "$UserAppDataDir\cura\$CuraVersionDir"
    RMDir /r "$UserLocalAppDataDir\cura\$CuraVersionDir"
  remove_user_dir_no:
```

diegopradogesto, 2018-11-12:
Devs. CURA-5927

antip8b, 2019-03-18:
Is the issue resolved for 3.6.0?

Ghostkeeper, 2019-03-19:
It was, until the certificate of the Arduino drivers installation expired.

jlawrence1984, 2019-06-12:
Any ETA when a silent uninstall might work without prompting for the config removal? Even is we could add a parameter to the uninstall.exe to do this

Ghostkeeper, 2019-06-13:
I believe this is currently planned for version 4.3. This issue has been brought up a lot by network administrators who want to roll out Cura on their business networks.

Ellecross, 2019-08-30:
This is fixed now and will be in Cura 4.3. 
The silent installation runs and the prompt to remove config does not show up anymore. This leaves the config in place.

htwsaaraub, 2019-10-28:
@Ellecross 
Today I also run in this issue. I confirm that the silent installation runs correct (Exit code 0).
Although after the uninstall process the Uninstall.exe is still in place. Because of this the Install Folder won't be deleted after uninstall. 
If you run the Uninstaller without silent paramter the Folder is deleted correctly. 
I use the following Command for Uninstall.

`{%SystemDrive%}\Program Files\Ultimaker Cura 4.3\Uninstall.exe /S _?={%SystemDrive%}\Program Files\Ultimaker Cura 4.3\`

damon-atkins, 2019-10-30:
Pls reopen given the above info

Ellecross, 2019-11-04:
Hey @htwsaaraub and @damon-atkins I did some more investigation. It looks like there is still instances of the Cura.exe running in the background (I think that was symptom of the [issue where Cura does not start consistently on Windows 65bit](https://community.ultimaker.com/topic/29861-cura-wont-open-in-windows-10-i-may-have-a-solution/?tab=comments#comment-249082) ).. 

After I killed the Cura.exes running in the background, I ran the uninstall.exe /S and the whole installation folder was deleted. Can you check if this was the same case for you? 

On that regard, should Cura then check first and then just kill the Cura.exes on silent uninstall? Something to discuss with the Cura team. ;) Thank you for reporting this!!! 

Ultimaker Cura devs, please see CURA-6953.

Ghostkeeper, 2019-11-04:
> On that regard, should Cura then check first and then just kill the Cura.exes on silent uninstall? Something to discuss with the Cura team. ;) Thank you for reporting this!!!

Not sure we can do something like this. There's a couple of problems with it:
* Applications don't normally have the rights to kill other applications on the operating system, unless they are launched by that application.
* We only configure the installer, we didn't program it, so executing arbitrary code is not always possible.

fieldOfView, 2019-11-04:
The uninstaller is executed "elevated" (with Admin privileges) and can kill processes. Here's how to do it with NSIS: https://stackoverflow.com/a/13023029

htwsaaraub, 2020-01-17:
@Ellecross 
This is also valid if Ultimaker was never started after a fresh Install.

@Ghostkeeper The Installer/Uninstaller Process is always run elevated under a specific User Account in Deployment Process. This User has Admin Privileges on all Computers, and could easyly kill other procesees. 

Just testet uninstall deployment of Version 4.4.1
The Uninstall.exe in the Cura Folder is still present. Folder wasn't deleted. 
This has nothing to do with my Deployment method. Even if I run this from a cmd command line the Uninstall.exe won't delete itself and the Folder. 

> "C:\Program Files\Ultimaker Cura 4.4\Uninstall.exe" /S _?=C:\Program Files\Ultimaker Cura 4.4\

Do you really force to delete all files AND explicit the uninstaller. 
I found an example for a NSIS uninstaller wich has a extra line for the uninstaller.

```
Section "Uninstall"
    Delete "$INSTDIR\*.*"
    Delete "$INSTDIR\Uninstall.exe"
    DeleteRegKey /ifempty HKCU "Software\APPName"
    RMDir /r "$INSTDIR"
SectionEnd
```

Ghostkeeper, 2020-01-20:
>     Delete "$INSTDIR\*.*"
    Delete "$INSTDIR\Uninstall.exe"

This doesn't make sense to me. `Uninstall.exe` also matches on `*.*`, right?

mahtDFR, 2020-01-21:
I was able to successfully replicate this. Reproduction steps:

* Installed 4.4.1 to `C:/Ultimaker Cura 4.4` (I changed from default directory to preserve my existing install)

* Ran command as administrator:
`$"C:\Ultimaker Cura 4.4\Uninstall.exe" /S _?=C:\Ultimaker Cura 4.4\`

Result: Ultimaker Cura 4.4 folder remains, containing `Uninstall.exe`

We'll discuss further

mahtDFR, 2020-01-24:
Devs: related to internal ticket CURA-7136

konskarm, 2020-02-11:
I'm closing this issue since it is fixed in 4.5-beta.
Re-open it if the issue persists in the beta release (scheduled for 11-Feb-2020) .

closed 2020-02-11 by konskarm