Ultimaker/Cura #4777: Z Offset Setting

nbystrom, 2018-11-11:
Cura Version 3.5.1
Windows 10

While configuring my Prusa i3 mk2, I placed painter's tape across the build plate, which raised the building surface by ~0.25 mm. This caused the first layer to print somewhat incorrectly, and as there was no setting in the settings menu that I could find, I was forced to introduce a z-offset using the start g-code, by using a G0 command to move the extruder to the right height, followed by a G92 command to zero the z axis at the offset. While this works, it is not a very efficient way of doing it, so I was hoping for a setting to be introduced which allows for the initial Z offset to be modified within the slicer.

smartavionics, 2018-11-11:
There's a plugin to do that, not tried it though.

[image]

diegopradogesto, 2018-11-11:
Yes, you can try this plugin out. You can find it in the Toolbox/Marketplace.

closed 2018-11-11 by diegopradogesto