Ultimaker/Cura #4778: [3.5.1] brim on the inner supports

mixtur, 2018-11-11:
**Application Version**
3.5.1

**Platform**
Windows 7

**Printer**
[CCR10_tail.zip](https://github.com/Ultimaker/Cura/files/2569287/CCR10_tail.zip)

**Actual Results**
[image]

**Expected results**
I would expect to see brim around that inner part of the support. It keeps unsticking.

alekseisasin, 2018-11-12:
After loading the project I see that you are using tree support, it is experimental and we cannot guaranty the correct output for all models. But we already had a fix/workaround for this case. You can wait for a new Cura 3.6 stable version or download 3.6 Beta version. In this version, you can find a setting "Enable Support Brim". After enabling the setting it, you will the brim around inner parts in the model.
For now, I will close the issue if you will still have the same issue in 3.6 version then we can re-open the issue.

closed 2018-11-12 by alekseisasin

mixtur, 2018-11-12:
The same happens with concentric supports, so I believe it is more general than that. I can't send you project right now but I will as soon as I can.

mixtur, 2018-11-12:
Ow. Sorry I didn't read the whole response. Ok. I'll wait for 3.6.

alekseisasin, 2018-11-13:
New Cura 3.6 is online. You can try it.