Ultimaker/Cura #4784: [3.5.1] Windows 10 font problem

uwezi, 2018-11-12:
**Application Version**
Cura 3.5.1 & 3.6

**Platform**
Windows 10 64 bit

**Steps to Reproduce**
start the application in non-administrator mode

**Actual Results**
text/numbers in most application settings fields is not visible
[image]

**Expected results**
a usable GUI

**Additional Information**
The issue is new, could be related to either a recent Windows Update/driver update. But I have to emphasize that the problem only exists when Cura is started from a regular user account. Starting Cura as Administrator works without problem.

Trying to give the user full access to the Cura installation directory does not help either, nor running it in diverse Windows compatibility settings.

alekseisasin, 2018-11-13:
It is strange, I assume you have some system configuration which does not allow to use fonts, You can check user privileges.

uwezi, 2018-11-15:
of all programs and scripts which I have running on this system only Cura is affected and only since quite recently. Not being allowed to use fonts sounds like a weird user privilege setting.

siehputz, 2018-11-27:
I have the same issue with the numbers not being visible, but for me running as Administrator made no difference. I am running Cura 3.6, 64 bit windows. I tried version 3.4; which had the same issue, and version 3.2.1 which was OK. Note that the numbers are still really there, and typing in the boxes, even tho you can't see what is being typed, does change the value of the variable. This only started after I upgraded from Windows 7 to Windows 10.

siehputz, 2018-11-28:
Did some more research. Is this possibly related to the Noto font issue reported at https://github.com/googlei18n/noto-fonts/issues/1046 ?

uwezi, 2018-11-28:
> 
> 
> Did some more research. Is this possibly related to the Noto font issue reported at [googlei18n/noto-fonts#1046](https://github.com/googlei18n/noto-fonts/issues/1046) ?

I don't have MS Office on my computer, but all Noto fonts are installed as fonts under Windows and work fine in LibreOffice on my system.

siehputz, 2018-11-29:
When I go to the Cura resources\themes\cura-light\fonts\noto-sans directory and click on noto-sans-400-regular.ttf and some others the font viewer shows an empty screen. But other fonts in the directory (such as  noto-sans-500-medium.ttf) look to be OK. This seems to be consistent with the reported issue where some of the weights don't work in some applications.

uwezi, 2018-11-30:
> When I go to the Cura resources\themes\cura-light\fonts\noto-sans directory and click on noto-sans-400-regular.ttf and some others the font viewer shows an empty screen. But other fonts in the [...]

You are completely right! There is something wrong with the display of the noto-sans-400 and Windows 10 on my system as well.

Now I wonder why it doesn't help to replace the 400-files with a copy of the 500-files, both in Cura's own directories and in the Windows font installations. Because now the "400"-font displays correctly, but still no numbers in Cura...

Next question: why does Cura have to supply its own fonts - aren't there enough fonts generally available on any modern system?

fieldOfView, 2018-11-30:
Here's a workaround, that selects the system default font instead of the (broken?) Noto Sans font:
* Go to Help -> Show configuration folder
* Navigate to "themes"
* Create a new folder named (eg) "default-font"
* In that folder, download this file (named theme.json):
https://gist.githubusercontent.com/fieldOfView/3e94440bcc66acfd014dd2f71f6a4885/raw/c097604673fa7745ae9668c69c1ac9d8c3f3919e/theme.json
* Restart Cura, and in the preferences set the theme to "Default font"
* Restart Cura again.

uwezi, 2018-11-30:
> .... 
>     * Restart Cura, and in the preferences set the theme to "Default font"
> 
>     * Restart Cura again.

Yes, thank you, this works fine for me!

fieldOfView, 2018-11-30:
Good to know the workaround works.

In order to figure out what is really going on, could you zip up the noto-sans-400 font file and attach it here? Just to rule out that eg a viruschecker modified/corrupted the font while installing Cura.

uwezi, 2018-11-30:
inspired by your new theme I did some modifications to the original theme.json and this also works for me:

``    "fonts": {
        "large": {
            "size": 1.35,
            "weight": 63,
            "family": "Noto Sans"
        },
        "large_nonbold": {
            "size": 1.35,
            "weight": 63,
            "family": "Noto Sans"
        },
        "medium": {
            "size": 1.16,
            "weight": 63,
            "family": "Noto Sans"
        },
        "medium_bold": {
            "size": 1.16,
            "weight": 63,
            "family": "Noto Sans"
        },
        "default": {
            "size": 1.0,
            "weight": 63,
            "family": "Noto Sans"
        },
        "default_bold": {
            "size": 1.0,
            "weight": 63,
            "family": "Noto Sans"
        },
        "default_italic": {
            "size": 1.15,
            "weight": 63,
            "italic": true,
            "family": "Noto Sans"
        },
        "small": {
            "size": 1.0,
            "weight": 63,
            "family": "Noto Sans"
        },
        "very_small": {
            "size": 1.0,
            "weight": 63,
            "family": "Noto Sans"
        },
        "button_tooltip": {
            "size": 1.0,
            "weight": 63,
            "family": "Noto Sans"
        },
        "setting_category": {
            "size": 1.15,
            "weight": 63,
            "family": "Noto Sans"
        },
        "action_button": {
            "size": 1.15,
            "weight": 63,
            "family": "Noto Sans"
        }
    },``

uwezi, 2018-11-30:
since I swapped the original font for the -500-font I'll end several files, but I think I can exclude some corruption of the file after installation....
[noto-sans.zip](https://github.com/Ultimaker/Cura/files/2634987/noto-sans.zip)
[noto-sans-display.zip](https://github.com/Ultimaker/Cura/files/2634990/noto-sans-display.zip)

siehputz, 2018-12-03:
Workaround worked for me as well, many thanks! I have attached the broken Noto 400 regular ttf file.
[noto-sans-400-regular.zip](https://github.com/Ultimaker/Cura/files/2639949/noto-sans-400-regular.zip)

fieldOfView, 2018-12-03:
If I open that font from your zip (by double clicking it), it seems to work fine (windows 10, 64 bit):
[image]

closed 2020-06-19 by nallath