Ultimaker/Cura #4785: 3.4.1 GUI - Free move on build plate unresponsive

bytte1, 2018-11-12:
**Application Version**
3.4.1

**Platform**
 Win 7

**Printer**

**Steps to Reproduce**
Load file using 3D print from Fusion 360. Object appears on build plate. Left click to select produces red blue green arrows, but more often than not holding left mouse button and dragging does NOT move the object.  After many repeated attempts it moves.

**Actual Results**
 

**Expected results**
 

**Additional Information**
Behavior is the same when opening Cura and loading stl from disk.  Move dragging the RGB move arrows works as expected, but dragging the object (free move) mostly does not.

Vandrasc, 2018-11-14:
Hi @bytte1 , we are not able to reproduce the issue as described by you. Are you very zoomed in or out when trying to drag the model? does this occurs with any model or a specific one? 

Could you try to zoom a bit closer towards the model and try to drag it then? Thank you!

bytte1, 2018-11-14:
This isn't just one model, although the ones I am building are all relatively small (20x15x15mm). The build plate is 220x220. Putting the model to the rear (using the arrows) seems to help. But it is still random whether any individual left click will grab the model or not. In all cases (positions/ zooms) success in grabbing is still less than 30%. Just because one move works, doesn't mean the next will. I can't associate the position of the click with likelihood of success.

zapta, 2019-04-10:
Same problem here, success in selection and moving is random and often fails. I need to click on different locations of the model trying to select it, often I need to try 3-5 times until success. Then moving sometimes work sometimes fails.

This is on macbook air with OSX 10.14.4, using the touch pad. This problem is specific to Cura. No similar select/drag issues with other apps on this computer.

nallath, 2019-04-11:
We recently improved this behavior, so it should be a lot better for 4.1

closed 2019-04-11 by nallath