Ultimaker/Cura #4786: cura 3.5.1 gives holes in the print where they should be closed.

khan04what, 2018-11-12:
**Application Version**
<Cura 3.5.1>

**Platform**
<WINDOWS 10 64 bit>

**Printer**
<ANYCUBIC I3 Mega>

**Steps to Reproduce**
<don't understand this question?>

**Actual Results**
<
- The result is thats Cura3.5.1 giving holes in the print where they should be closed. and I print these when saved as g-code file on sd-card and the print has than other place on the print more holes or little lines that not printed. >

**Expected results**
<This fixed hole and line bug>

**Additional Information**
<Everytime a open an stl file go to layer view and view the file to print also fixed print leave holes behind.>

smartavionics, 2018-11-12:
Please zip the project file and attach to this issue. Without that it's impossible to investigate - thanks.

khan04what, 2018-11-13:
http://www.thingiverse.com/thing:1229240
Zombie Cyclist (ZombWar) by Deividcole 

I tried to send my stl file via 7zip but that did not work, he did not accept it.
i have picture: 
[image]

[image]

maybe this will help you?

smartavionics, 2018-11-13:
Not really, I needed the project file to be able to check your settings. Anyway, it looks like you have coasting enabled. Try disablng that.

khan04what, 2018-11-13:
I disabled coasting it looks better in layer view now. I want to say that I had no problem with previous versions and certainly because the function had to reduce stringing. but I will also try another print. but for now in cura the print has no holes and looks oke.

rtorresca, 2018-11-18:
Same problem in 3.6.
Disabling coasting also works.

nallath, 2018-11-22:
This is intentional; Coasting is useful if your extruder needs some time to stop extruding. Even though that layer view shows that there is a hole, it does not mean there actually will be a hole.

closed 2018-11-22 by nallath