Ultimaker/Cura #4791: [3.6.0] slicer does not add layer to touch the bed

momala454, 2018-11-13:
**Application Version**3.6.0

**Platform**
windows 10

**Printer**
creality ender 3

**Steps to Reproduce**

**Actual Results**

**Expected results**

i'm using the support and adherence, but using the attached  stl file, it looks like some of the model does not touch the bed, and even with support and adherence, cura does nothing to make it touch the bed, so when i try to print, when it arrives at the second layer to the point where it tries to print the element that does not touch to bed, it print in the air and does not stick at all to the bed.

**Additional Information**

[image]
[image]

[o_Door_Side_Top.zip](https://github.com/Ultimaker/Cura/files/2577599/o_Door_Side_Top.zip)

momala454, 2018-11-13:
yes that's what i understood, but generating "support" or "adherence", should make something to fix this, no ?

smartavionics, 2018-11-13:
Yes, in general, it does. But in this case the gap is smaller than the layer height so it won't possibly put support in there.

momala454, 2018-11-13:
my layer height is 0.1

smartavionics, 2018-11-13:
OK, but, remember, you also need at least one layer of air above the support otherwise it will never separate from the part being supported. If you really want support under that model you need to raise it sufficiently. Here I raised it to 0.5mm and it put support under the whole model...

[image]

momala454, 2018-11-13:
so no "automatic" solution in cura, i have to manually verify that the object is not floating and if it does, put it underground or above ?

smartavionics, 2018-11-13:
Correct. Goodbye.

momala454, 2018-11-13:
can you consider adding a feature that detect this kind of problem and warn or automatically find solutions ?

alekseisasin, 2018-11-14:
Hi @momala454 , I think it is out of the Cura scope. It is normal practice before sending Gcode to a printer user needs to ensure that it model is correctly prepared. "Detect this kind of problem" - this is a feature request. If you have a clear idea of how it should work, you can always make a pull request and even propose your solution.
If you do not mind to close the issue, because @smartavionics answered your question.

closed 2018-11-14 by momala454