Ultimaker/Cura #4793: [BUG][3.6] Custom Printer Filament diameter reset every time to 2.85

gianmarcov, 2018-11-14:
**Application Version**
3.6

**Platform**
Mac OS X Sierra 10.12 64 Bit on MacBook Pro Retina

**Printer**
Alfawise U10 (Ramps Plus2 BT7200)

**Steps to Reproduce**
Start Cura -> click on printer -> add printer -> custom 'Custom FFF Printer ' -> 
set H 500 L 400 W 400 
set Nozzle Diameter 0.4
set  Filament diameter 1.75

After closing the window and edit printer settings again it shows 2.85.

**Actual Results**
Cannot set filament diameter to 1.75, supported filament shows  2.85 when changing to 1.75 will cause
error message 'The given filament diameter is not supported by extruder'.
The printing has large gaps between printed lines, the printed abs object is flexible, has fiber surface.

**Expected results**
Should maintain the value to 1.75 and show 1.75 as supported filament.

**Additional Information**
Workaround is to use 'Creality cr-10s' as printer and change print dimensions to H 500 L 400 W 400 in the printer settings. It will print without gaps and option usb printing will be accessible (Works like a charm).

Vandrasc, 2018-11-14:
Hi @gianmarcov, It shows 1.75 to me, even after the restart. Could you please share the project file with us so we can see what the issue could be? Thank you!

deacon240, 2019-01-28:
This doesn't happen for me every time but after configuring a new custom printer I had the same issue. After spending two days troubleshooting extrusion and various other bits of hardware I restarted Cura and what do you know? It reverted to 2.85 and never saved it. This has been an ongoing bug in Cura for at least 2 years. 

It seems to happen most often with custom printers and it appears to not always save the changes in the configuration upon closing of the machine settings window.

photodude, 2019-09-07:
Just imported the json profiles from BIBO MFG on Cura 4.2.1 and although the json is 1.75 they imported at 2.85mm.
Seems that the fix #2300 did not solve this issue.  json files from MFG included (note: the MFG forgot to specify the MFG name in the files so they show up under the "na" portion of the printer list)

[BIBO2 Cura settings.zip](https://github.com/Ultimaker/Cura/files/3587155/BIBO2.Cura.settings.zip)

Ghostkeeper, 2019-09-09:
In BIBO's case that is correct because the 2.85 originates from the BIBO1.def.json extruder definition which specifies that it inherits from fdmextruder.def.json which says 2.85mm.

The compatible material diameter is defined per extruder, so if BIBO needs 1.75mm materials it needs to specify that in its extruder definitions.

Changing the diameter and restarting Cura, it seems to properly save the new diameter.

photodude, 2019-09-10:
@Ghostkeeper interesting that  fdmextruder.def.json material default is 2.85mm.
I've now modified all of my BIBO extruder json definitions to include the following overrides
```json
"material_diameter": {
    "default_value": 1.75
},
"machine_nozzle_size": {
    "default_value": 0.4
},
```
Thanks for pointing me in the right direction on why I was seeing 2.85 and how to correct the json def with the overrides.

closed 2020-06-19 by no-response[bot]