Ultimaker/Cura #4795: [3.6] Slice Issues - broken model lines when slicing.

ncwhale, 2018-11-14:
**Application Version**
3.6

**Platform**
WIndows 10 64bit with i7 4770 / 32G MEM

**Printer**
Anycube Chiron
[PI3_Midchildan system(v2).zip](https://github.com/Ultimaker/Cura/files/2581643/PI3_Midchildan.system.v2.zip)

**Steps to Reproduce**
Open Cura -> Import my model -> Display slice view.

**Actual Results**
Slice broken model lines which can be printed.
[image]

**Expected results**
Let the model be printed use one or two lines.

**Additional Information**
I disabled support to let the capture image more clear.

smartavionics, 2018-11-14:
The model has problems.

[image]

ncwhale, 2018-11-15:
What's this tool's name?

smartavionics, 2018-11-15:
It's the mesh tools plugin.

[image]

Liger0, 2018-11-26:
> 
> 
> What's this tool's name?
> 
> Oh, I See. Any tools suggest to fix this model? I made it by tinkdesk...

You could try windows 3d builder for fast automatic fixes.

ncwhale, 2018-12-03:
I tried almost every tool can find to fix the .STL file, and still, got the Gcode generate error: some lines are missing.

closed 2020-11-03 by nallath