Ultimaker/Cura #4801: Gray surrounding volume incorrectly calculated when printing One at a Time

MarcusAdams, 2018-11-15:
**Application Version**

3.6

**Platform**

Windows 10 x64

**Printer**

Custom Prusa MK2

**Steps to Reproduce**

Add model to print.
Change print sequence: Special Modes->Print Sequence->One at a Time

**Actual Results**

Gray box surrounding part is larger than it's supposed to be and larger than previous version (3.5). This prevents you from printing more than one part "One at a Time" and won't even let you slice since Cura thinks there's no room on the plate.

**Expected results**

Gray area surrounding part should be the correct size based on printer settings so that you can load more than one part.

**Additional Information**

It seems that there is a workaround, which is to save the .3mf in the previous version, load the .3mf into 3.6, then you can slice it as long as you don't change any settings.

alekseisasin, 2018-11-15:
Hi @MarcusAdams , indeed it has a different gray area. The reason is that in previous Cura version this area was calculated not properly, and profiles also could have incorrect/missing data. This is the case of this issue. I would like to ask to help us with this profile.
Can you check the printer profile:
https://github.com/Ultimaker/Cura/blob/master/resources/definitions/prusa_i3_mk2.def.json

For calculating the gray area we need two settings:
1) machine_head_polygon
2) machine_head_with_fans_polygon

We cannot change them without having them the printer, because they depend on the technical 3D printer (Prusa) specs. It is the dimension of the print head.
If you can adjust them (You can see how they defined for Ultimaker printer) according to your 3D printer dimensions then we can add them to the next Cura release as well. Both of these settings should present.

MarcusAdams, 2018-11-15:
I don't understand because these values are set in Printhead Settings. Here are my settings. That gray shape is WAY too big.

[image]
[image]

MarcusAdams, 2018-11-15:
I don't understand because this is set in the printer setup under Printhead Settings. Looking at it closer, the gray shape seems to be twice as big as it's supposed to be.

Here are my settings and how it looks in Cura:
[image]
[image]

alekseisasin, 2018-11-16:
Now I see, it is indeed an issue. I have checked similar issues and we already have a ticket for it. It is already in our Sprint. 
DEV: CURA-5822

MarcusAdams, 2018-11-16:
Thanks. I'll close this issue now.

closed 2018-11-16 by MarcusAdams

JeremyDWilliams, 2019-04-16:
This has popped up again in 4.0.0

Ghostkeeper, 2019-04-17:
Let's keep the discussion in here then: https://github.com/Ultimaker/Cura/issues/4720