Ultimaker/Cura #4804: Load gcode broken

BagelOrb, 2018-11-15:
**Application Version**
3.6 (wasn't a bug in 3.5.x)

**Platform**
Linux Mint

**Printer**
UM3

**Steps to Reproduce**
Drag and drop gcode file
[cube.gcode.zip](https://github.com/Ultimaker/Cura/files/2584456/cube.gcode.zip)

**Actual Results**
loading bar completes
switch to layer view
Nothing happens after that

**Expected results**
gcode is shown
time and material estimates are present
I can send the gcode to my printer over the network

**Additional Information**
Log:
```
2018-11-15 10:09:37,971 - DEBUG - [Thread-3] GCodeReader.FlavorParser.processGCodeStream [292]: Preparing to load GCode
2018-11-15 10:09:37,977 - DEBUG - [Thread-3] UM.Settings.ContainerRegistry.addContainer [371]: Container [per_object_stack_140196725169080] added.
2018-11-15 10:09:37,981 - DEBUG - [Thread-3] GCodeReader.FlavorParser.processGCodeStream [332]: Parsing Gcode...
2018-11-15 10:09:38,187 - DEBUG - [Thread-3] GCodeReader.FlavorParser.processGCodeStream [455]: Finished parsing Gcode
2018-11-15 10:09:38,189 - DEBUG - [Thread-3] GCodeReader.FlavorParser.processGCodeStream [467]: GCode loading finished
2018-11-15 10:09:38,193 - DEBUG - [Thread-3] UM.FileHandler.ReadFileJob.run [65]: Loading file took 0.2 seconds
2018-11-15 10:09:38,209 - INFO - [MainThread] SimulationView.SimulationView.calculateMaxLayers [375]: Min thickness can't be calculated because all the values are zero
2018-11-15 10:09:38,217 - INFO - [MainThread] SimulationView.SimulationView.calculateMaxLayers [375]: Min thickness can't be calculated because all the values are zero
2018-11-15 10:09:38,234 - DEBUG - [MainThread] UM.Controller.setActiveView [92]: Setting active view to SimulationView
```

[image]

alekseisasin, 2018-11-15:
I have opened the attached gcode on Linux, in Cura 3.6. It works

Vandrasc, 2018-11-15:
Works for me as well!