Ultimaker/Cura #4805: Missing features

SirBastol, 2018-11-15:
there is missing features .. 

1. when printing it should show tool path and current layer like repetier host
2. it shoudl show graphs of current travel speeds and heat monitoring..  (like repteier host)
3.  keep logs of your prints and the gcode files related to them in a database. so you can go back and 'reprint' or build profiles from those past builds. 
4.  store settings in cloud.. - custom prints/ bed layouts  etc. 
5.  when usb printing  . allow for timelapses with a webcam as you print  - options to save last frames  when you abort a print to document failures and why they failed ( adhesion for example .. and then  perhaps suggest possible fixes. 
6. a printer tuning wizard - for calbrating extruders steps and motor movemnt steps and such .
7. link to repositories with firmware versions for your printers mainboard so you can update
8. a setting page on printer for customs to chose stepper drivers./ mainboard model and  stepper motor models ( for further tuning algorithms from learning form the database
9. advanced idea : use webcams to not jsut do timelapses but perhaps photo recognition algorythms to detect if print is matchign expected bed output.. ( and a side by side comparason  between tool paths and the webcam feed.  so you can see that things are as expected)
10 . more accurate shapes for the  filament srawing in layer view ( ie squashed not round)
11. ability to print to multiple printers  and monitor them as above  
12. ability to controll a print farm or  monitor a print in progress that was goign on a usb sd card upload. 
13. ways to multiprocess a slice.. ( some places need support some dont.. some need infill some dont. .   
14. infil that goes automatically to freate the max angle for not needing supports.. instead of always vertical to bed..  create a support to the part at 70 degrees if thats what max angle is for unsupported.. 
15. if angle gets steep  walls shoudl automatically thicken to give it more ability to not need supports
16. keep a repository of settings in a database . that can be 'ultimaker/cura  pro profiles for printers and or enable access to public createed shared repositories of profiles in teh web/cloud  that are built based on your hardware config/motherboard etc.. 
17.. ability to do manual gcode on say a direct connection .. for diagnostics.. 
18 visual bed levelling/ and mesh bed creations visualization 
19. manual control wizard  making mesh bed  with many points and saving it to adjsut gcode slicing .. so you do it once and then  the printer does t have to do it every time
its built in  ( ie bed is warped but doesnt shapechange.. so no need to remesh every print)

smartavionics, 2018-11-15:
Good job that Cura is open source, eh? There's nothing stopping you adding the above features.

SirBastol, 2018-11-15:
true  but when i work 18 hour days on a set and my home time is filled with
printing and family theres no room for me to learn to make a cura plugin.
sometimes creative people need to be inspired with a wanted list to give
them ideas..

i wanted to inspire improvements ..
cause right now when i print with cura it looks like this.. which says
nothing  about whats supposed to be goign on . and it shoudl show more
'stats' when live printing

[image: image.png]

and that data can be saved and used to  make better one button profiles..

3d printing is not takign off because people are lazy and not everyone is a
tinkerer and a coder and an artist and a 3d design and cad modeller  all in
one.. and the ones that are  enjoy tinkering enough they just get it done
instead of trying to make things simpler and more self sufficient

Paul Cole
http://ca.linkedin.com/in/paulcolevfx
--=-=-=-=-=-=--
http://db.tt/LXIuAye0
^^click me for free cloud storage^^

“Defeat is a state of mind. No one is ever defeated until defeat has been
accepted as reality. "
- Bruce Lee
"http://tools.ip2location.com/ib2"  <http://tools.ip2location.com/ib2>

ChrisTerBeke, 2018-11-15:
Number 4 and 15 are actually something we're working on. The Backups plugin in the Marketplace is a first step towards those.

smartavionics, 2018-11-15:
Just kidding!

SirBastol, 2018-11-15:
lol

all good brotha. if i had the time or was getting paid i woudl totally do
it.
but i pritn on ashoestring budget from my underpaid job and overpriced rent
in vancouver bc,,
which means work allthe time or starve/homeless

Paul Cole
http://ca.linkedin.com/in/paulcolevfx
--=-=-=-=-=-=--
http://db.tt/LXIuAye0
^^click me for free cloud storage^^

“Defeat is a state of mind. No one is ever defeated until defeat has been
accepted as reality. "
- Bruce Lee
"http://tools.ip2location.com/ib2"  <http://tools.ip2location.com/ib2>

SirBastol, 2018-11-27:
> I think 3 is quite important for me and a lot of people besides me: saving all printed gcodes to a database automatically so that I can later look and see which gcode exactly printed.

The more data collected the more accurate and more details solutions can be sought

Having a tuning wizard that does things like walk you through getting the bed level manually and manual mesh bed levelling and running simple temp towers and retraction towers when you change filaments.  To funky tune the settings automatically as opposed to guessing and learning is much better than guessing. And wasting prints 

This includes tuning e steps and extruder steps etc 

It’s nice to have teoubleshooters but better to have software ai with data logging and sharing on fail and why it failed will improve the process for alll 

Sent from my iPhone please excuse typo mess or voice to text errors.

Ghostkeeper, 2020-04-21:
Closing this as duplicate of https://github.com/Ultimaker/Cura/issues/6020. If you meant to create feature requests for more than one feature, please create separate tickets. Otherwise this can never be closed.

closed 2020-04-21 by Ghostkeeper