Ultimaker/Cura #4806: Top View not above bed center [Any version].

Thisismydigitalself, 2018-11-15:
To begin with, Thank you for Cura 3.6. so far looking good.

**Application Version**
3.6 or any other recent version

**Platform**
Win10/64

**Printer**
CFFFP, Delta & Cartesian

**Steps to Reproduce**
click on top-view

**Actual Results**
Camera never directly above center of bed which makes it a bit tricky to place multiple objects close to bed edge.

**Expected results**
Not sure.. i always expect to see the camera diretly above center of bed, preferbly in orthographic view.

**Additional Information**
Just wondering what one should expect. i think i prefer Top orthographic view.

[image]
[image]
[image]
[image]

Ghostkeeper, 2019-07-17:
Fixed in version 4.2.

Ghostkeeper, 2019-07-17:
You can't move the camera 100% to the top view because of the gimbal lock. Clicking on the button now sets it to the exact top view, but if you move the camera it'll reset to a 0.1 radian angle from the top.

closed 2019-07-17 by Ghostkeeper