Ultimaker/Cura #4809: No Z-HOP when retracting for combed moves

notnyt, 2018-11-15:
**Application Version**
3.5.1

**Platform**
win

**Printer**
custom

**Steps to Reproduce**
print part with infill, comb: all, and z-hop when retract set

**Actual Results**
z-hop when retracted indicates it will z-hop any time there is a retraction.  It does not, but it should.  This is good to prevent the nozzle from impacting the print

[image]

no z-hop while moving to print infill
````
G1 F6000 E-0.3
G0 F18000 X175.82 Y101.971
G1 F3600 E0.3
````

**Expected results**
z-hop should be performed when retracting

**Additional Information**

[image]

````

Vlad-bi, 2018-12-25:
I have the same problem. What is a solution?

Vandrasc, 2019-01-03:
devs CURA-6072

Vlad-bi, 2019-01-03:
> devs CURA-6072
Don`t understand you. Is it cura version or id of issue?

Ghostkeeper, 2019-01-04:
ID of issue in Jira, the software that the Cura team uses to plan stuff and prioritise. Our Jira is not public though, because it also contains plans for supporting Ultimaker's future products which must be kept secret until release.

Vlad-bi, 2019-01-06:
I print with first and second extruder, but the problem did not disappear.

>using the buttons on the left of the screen.

Where is this buttons(in cura, in 3d printer)

> devs CURA-6072

And How this can help me(may i don`t understand something, sorry)

Vandrasc, 2019-01-14:
Hi @Vlad-bi,

"And How this can help me(may i don`t understand something, sorry)?" - This is only for our internal use, this way our developers know that we have created a ticket and now linked with your bug report, it helps us tracking bugs and it is for our administration. 
The issue ID will not help you in any way, except that you then know that we do have an eye on the issue, it is just our work process.

Vlad-bi, 2019-01-14:
Hi, @Vandrasc
So I should wait until this function appears in future versions of the program, shouldn`t it?

Vandrasc, 2019-01-28:
I had a look at the z-hop when printing multiple cubes for examples, and it behaves as intended, I cannot reproduce the issues mentioned above. 
In the project file, the dice is printed with extruder 1 but ALL of the extruder settings are set to extruder 2, causing the infill, walls, skin, etc. to print with extruder 2 and thus effectively printing the die with extruder 2. However, the Z Hop height setting is still taken from extruder 1 for moves within this model from walls to infill and such. You could see this as a bug but currently, it's been expected behavior, since the travel move was not part of any particular feature but only part of the model, so it should use the model's settings. The project file attached to this ticket has a z-hop of 0.1 mm and the combing is on, if you always want z-hop than combing should be off, there will not be a z-hop when combing unless it travels between parts. I hope this makes sense. 

We did cover a bug where the Z-hop could be missing if hopping during a layer change if the "Retract at Layer Change" setting was enabled, that is fixed now. 

We will close it for now but if you have different scenarios to reproduce the missing z-hop then please feel free to re-open. 

Thank you for your feedback!

closed 2019-01-28 by Vandrasc