Ultimaker/Cura #4813: The machine settings window is too short at the start and need to be stretched everytime.

a4jp-com, 2018-11-16:
**Application Version**

Version 3.6.0
**Platform**

Windows 10 64 bit
**Printer**

Dreammaker Overlord Pro
**Steps to Reproduce**

Just open machine settings in the settings area
**Actual Results**

popup is too small
**Expected results**

The popup should be a bit longer so we don't have to stretch it everytime
**Additional Information**

alekseisasin, 2018-11-20:
Hi @a4jp-com , On of the possible solutions is to store the machine settings pop up window size in cura.cfg. Another side of this request Cura team does not have much time for it.

a4jp-com, 2018-11-21:
Can you tell me where the code is and I'll edit the code and create a push if that works?

alekseisasin, 2018-11-22:
You might check how Cura stores/updates the "window_width" in file cura.cfg. The file is located here:
https://github.com/Ultimaker/Uranium/blob/master/UM/Qt/Bindings/MainWindow.py

GregValiant, 2024-11-24:
@a4jp-com I'm cleaning house around here.  This has been open for several years.  Can I close this?

closed 2024-12-09 by github-actions[bot]