Ultimaker/Cura #4814: [3.5.1] {speed_print} replacement in printer definition

gosseadema, 2018-11-16:
**Application Version**

Version 3.5.1

**Platform**

Windows 10

**Printer**

Anet A8: Created my own resources files for the Anet A8.

**Steps to Reproduce**

Select Anet A8 printer (anet_a8.def). 
Select a STL file.
Generate and save Gcode file.

**Actual Results**

The header of my printer definition file:

;-- Gosse Adema -- \n
;Sliced at: {day} {date} {time} \n
;Basic settings: Layer height: {layer_height} Walls: {wall_thickness} Fill: {infill_sparse_density} \n
;Print time: {print_time} \n
;Filament used: {filament_amount}m {filament_weight}g \n
;Filament cost: {filament_cost} \n
;M190 S{print_bed_temperature} 
;Uncomment to add your own bed temperature line \n
;M109 S{print_temperature} ;Uncomment to add your own temperature line \n
G21                     ;metric values \n
G90                     ;absolute positioning \n
M82                     ;set extruder to absolute mode \n
M107                    ;start with the fan off \n
G28 X0 Y0               ;move X/Y to min endstops \n
G28 Z0                  ;move Z to min endstops \n
G1 Z15.0 F{speed_print} ;move the platform down 15mm \n
G92 E0                  ;zero the extruded length \n
G92 E0                  ;zero the extruded length again \
nG1 F{speed_print} \nM117 Printing...  \n;--"

This is translated to:

;-- Gosse Adema --
;Sliced at: Fri 16-11-2018 09:59:20
;Basic settings: Layer height: 0.2 Walls: 0.8 Fill: 20
;Print time: 02:29:16
;Filament used: [4.61]m [12.19428310546875]g
;Filament cost: [0.30485707763671877]
;M190 S100.0 ;Uncomment to add your own bed temperature line
;M109 S230 ;Uncomment to add your own temperature line
G21                     ;metric values
G90                     ;absolute positioning
M82                     ;set extruder to absolute mode
M107                    ;start with the fan off
G28 X0 Y0               ;move X/Y to min endstops
G28 Z0                  ;move Z to min endstops
G1 Z15.0 F30 ;move the platform down 15mm
G92 E0                  ;zero the extruded length
;G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F30
M117 Printing...

**Expected results**

G1 Z15.0 F{speed_print} ;move the platform down 15mm \n
and
G1 F{speed_print} \nM117 Printing...  \n;--"

is translated to:

G1 Z15.0 F30 ;move the platform down 15mm
and
G1 F30

This looks correct, but it should be

G1 Z15.0 F1800 ;move the platform down 15mm
and
G1 F1800

The speed_print value is 30 mm/sec. This is the value from the defenition file and cura.
But the GCODE file values are in mm/min. 

(Workaround: Use a fixed/hardcode value in the header.)

**Additional Information**

[resources.zip](https://github.com/Ultimaker/Cura/files/2588624/resources.zip)

alekseisasin, 2018-11-19:
What do you mean  "Created my own resources files for the Anet A8"
If Cura does not have Anet A8 definitions, then it is hard to check your issue. First should be checked the definitions. Might be the issue is there.

fieldOfView, 2018-11-19:
The issue is the same with any printer definition; Speeds are entered as mm/s in Cura, but gcode expects mm/min. So if you use a speed in a replacement pattern, it will not be correct for gcode.

gosseadema, 2018-11-19:
@alekseisasin 

I've added my Anet A8 configuration files in the "resources.zip"-file.

@fieldOfView

I'll use the workaround, and use fixed values in the Gcode Header.

closed 2019-03-01 by nallath