Ultimaker/Cura #4820: Cura 3.6 (but also 3.5.x) extremely slow on MacBook Air I5 4GB 2011

pieri70, 2018-11-17:
starting from cura 3.5.x onward cura is not usable on my macbook air 2011.
The mac has just been freshly reinstalled with Ma OS High sierra (it was the same with el capitain).
It's I5 1,7Ghz CPU, 4GB RAM, SSD 256GB.

Cura start normally but then it's near unresponsive.
Any action that I take with program bust CPU >100%.
Changing options takes a lot of time.
Layer view is almost unresponsive, I cannot move the layer selection.

Cura 3.4.1 works fine

Here is the log file

`2018-11-17 07:48:42,590 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FileLogger
2018-11-17 07:48:42,609 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin LegacyProfileReader
2018-11-17 07:48:42,627 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SimpleView
2018-11-17 07:48:42,645 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SolidView
2018-11-17 07:48:42,666 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin FirmwareUpdateChecker
2018-11-17 07:48:42,687 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin SupportEraser
2018-11-17 07:48:42,704 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [422]: Loaded plugin PrepareStage
2018-11-17 07:48:42,741 - INFO - [MainThread] UM.VersionUpgradeManager.upgrade [130]: Looking for old configuration files to upgrade.
2018-11-17 07:48:42,844 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./packages.json, skip it
2018-11-17 07:48:42,847 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./plugins.json, skip it
2018-11-17 07:48:42,856 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker2_extended, skip it
2018-11-17 07:48:42,860 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_7, skip it
2018-11-17 07:48:42,864 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_1, skip it
2018-11-17 07:48:42,868 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_6, skip it
2018-11-17 07:48:42,872 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_8, skip it
2018-11-17 07:48:42,877 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker2_extended_extruder_0, skip it
2018-11-17 07:48:42,883 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/fdmprinter, skip it
2018-11-17 07:48:42,889 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker_s5, skip it
2018-11-17 07:48:42,896 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker2, skip it
2018-11-17 07:48:42,902 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_3, skip it
2018-11-17 07:48:42,908 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_4, skip it
2018-11-17 07:48:42,912 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_5, skip it
2018-11-17 07:48:42,918 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_2, skip it
2018-11-17 07:48:42,927 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom, skip it
2018-11-17 07:48:42,954 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./packages.json, skip it
2018-11-17 07:48:42,959 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./plugins.json, skip it
2018-11-17 07:48:42,967 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker2_extended, skip it
2018-11-17 07:48:42,973 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_7, skip it
2018-11-17 07:48:42,978 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_1, skip it
2018-11-17 07:48:42,982 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_6, skip it
2018-11-17 07:48:42,987 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_8, skip it
2018-11-17 07:48:42,992 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker2_extended_extruder_0, skip it
2018-11-17 07:48:42,996 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/fdmprinter, skip it
2018-11-17 07:48:43,002 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker_s5, skip it
2018-11-17 07:48:43,006 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker2, skip it
2018-11-17 07:48:43,009 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_3, skip it
2018-11-17 07:48:43,013 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_4, skip it
2018-11-17 07:48:43,016 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_5, skip it
2018-11-17 07:48:43,020 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_2, skip it
2018-11-17 07:48:43,023 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom, skip it
2018-11-17 07:48:43,043 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./packages.json, skip it
2018-11-17 07:48:43,046 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: ./plugins.json, skip it
2018-11-17 07:48:43,050 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker2_extended, skip it
2018-11-17 07:48:43,053 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_7, skip it
2018-11-17 07:48:43,056 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_1, skip it
2018-11-17 07:48:43,061 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_6, skip it
2018-11-17 07:48:43,066 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/custom_extruder_8, skip it
2018-11-17 07:48:43,070 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/ultimaker2_extended_extruder_0, skip it
2018-11-17 07:48:43,073 - WARNING - [MainThread] UM.VersionUpgradeManager._getUpgradeTasks [281]: Failed to get file version: cache/definitions/3.6.0/fdmprinter, skip it
[... cut ...]
2018-11-17 07:48:56,322 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura3.6.app/Contents/MacOS/resources/shaders/grid.shader]...
2018-11-17 07:48:56,331 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura3.6.app/Contents/MacOS/resources/shaders/platform.shader]...
2018-11-17 07:48:58,118 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [181]: Error check finished, result = False, time = 3.4s
2018-11-17 07:49:06,929 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
`

alekseisasin, 2018-11-19:
@LipuFei  maybe you can check the error.

alekseisasin, 2018-11-21:
I assume the problem is related to a new QT framework. It was changed from 5.8 to 5.10 version. After it, "old" OS system could not support certain features.
I do not have Macbook and do not have a lot of experience with it, is it possible to update OS? (like install new windows version)

LipuFei, 2018-11-21:
Hi @pieri70 , I cannot reproduce the high CPU usage problem unfortunately, but we are planning to make some tool to do CPU usage profiling in Cura so that users who have this issue can help us investigating those problems. We will keep you updated.

pieri70, 2018-11-21:
Hello @LipuFei thanks for your help.
Let me know if I can submit some other info.
@alekseisasin I just installed on a clean (formatted) hard disk the second-last version of OS, the last one usable on my hardware..
Cheers 
Pietro

app-js, 2018-12-17:
+1 Since version 3.5 it is no longer possible to work with it.

nean-and-i, 2019-02-02:
+1 unusable slow on macpro i7, 32GB Ram , Nvidia GTX 1070
not a ram issue or hw issue.it consumes at min. twice as much ram as simplify3d.
Slic3r PE and Simplify3D  run in lightspeed compared to cura.

pieri70, 2019-02-02:
Same with beta 4, slightly better with "force layer compatibility mode" in preference. But still unusable.

fieldOfView, 2019-02-03:
The problem is that Cura is using a software renderer, instead of hardware accelerated rendering of the 3d scene. 

Your GPU/driver may not support OpenGl 4.1 (which is used for the advanced parts of the layer view), and instead of falling back to OpenGl 2 (which it did in previous versions), it uses software rendering to meet with the requested OpenGl 4.1 features.

nean-and-i, 2019-02-09:
> The problem is that Cura is using a software renderer, instead of hardware accelerated rendering of the 3d scene.
>
> Your GPU/driver may not support OpenGl 4.1 (which is used for the advanced parts of [...]

My driver appears to support it, according to OpenGL Extension Viewer, but nevertheless it's slow as hell.
Is there a way to debug it?
Would it be possible to choose a more common/compatible openGL render or at least provide an option to switch to another one?

[image]

fieldOfView, 2019-02-09:
@pieri70 shared logs, so I could see that his Cura is using software rendering. Can you share yours?

nean-and-i, 2019-02-09:
sure, I guess you're looking for something like this:

`9-02-09 21:27:50,770 - DEBUG - [MainThread] UM.Qt.QtRenderer._initialize [183]: Support for Vertex Array Objects: True
2019-02-09 21:27:50,788 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [99]: Initialized OpenGL subsystems.
2019-02-09 21:27:50,791 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [100]: OpenGL Version:  4.1 NVIDIA-10.33.0 387.10.10.10.40.113
2019-02-09 21:27:50,795 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [101]: OpenGL Vendor:   NVIDIA Corporation
2019-02-09 21:27:50,798 - DEBUG - [MainThread] UM.View.GL.OpenGL.__init__ [102]: OpenGL Renderer: NVIDIA GeForce GTX 1070 OpenGL Engine
2019-02-09 21:27:50,802 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/default.shader]...
2019-02-09 21:27:50,810 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/selection.shader]...
2019-02-09 21:27:50,817 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/default.shader]...
2019-02-09 21:27:50,823 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/composite.shader]...
2019-02-09 21:27:50,833 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/overhang.shader]...
2019-02-09 21:27:50,843 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/striped.shader]...
2019-02-09 21:27:50,852 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/transparent_object.shader]...
2019-02-09 21:27:50,860 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/striped.shader]...
2019-02-09 21:27:50,887 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/default.shader]...
2019-02-09 21:27:50,894 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/grid.shader]...
2019-02-09 21:27:50,902 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [/Applications/Ultimaker Cura.app/Contents/MacOS/resources/shaders/platform.shader]...`

fieldOfView, 2019-02-10:
@nean-and-i it looks like yours is a different issue than @pieri70's. I'm guessing that in your case, rotating the viewport and scrolling through the layers in layerview isn't all that bad, but you are mainly having issues when changing values and interacting with the sidebar, right?

nean-and-i, 2019-02-10:
> @nean-and-i it looks like yours is a different issue than @pieri70's. I'm guessing that in your case, rotating the viewport and scrolling through the layers in layerview isn't all that bad, but you are mainly having issues when changing values and interacting with the sidebar, right?

That's about right.
I tested to remove the printer (cfdm) and add one from the list (Tevo Tornado) and it appears to be faster now (1-2secs).
however, it looks like cura is checking against the definition after each click if an option is valid? Why such check can take so long and isn't there a more efficient way for that?

pieri70, 2019-02-10:
Here is again my cura log if can be useful
I am not skilled in reading it but at the end I can see many opengl messages..
I'm in compatibility mode, high sierra, cura 3.6

[cura.log](https://github.com/Ultimaker/Cura/files/2849309/cura.log)

pieri70, 2019-02-10:
and this is cura 4.0 beta log
[cura.log](https://github.com/Ultimaker/Cura/files/2849313/cura.log)

Ghostkeeper, 2021-05-13:
How is this performing in Cura 4.9? We've updated the Qt version, which affects a lot of these things.

closed 2021-05-27 by no-response[bot]