Ultimaker/Cura #4823: [3.6.0] Duplicate print jobs through Octoprint plugin

jarrodasmith, 2018-11-17:
**Application Version**
3.6.0

**Platform**
Windows 10

**Printer**
CR-10, Ender2

**Steps to Reproduce**
Install (in my case upgrade) the Octoprint plugin for Cura 3.6.  Connect Cura to the Octoprint instance attached to your printer.  Slice a file and choose "Print to Octoprint".

**Actual Results**
The gcode runs fine.  But sometimes (as of this writing I don't know what I'm doing differently these times) as soon as the print finishes, the print job gets sent *again* without user intervention.  The second instance appears to start at X,Y = 0,0, NOT where the GCode had it placed.  This has occurred at least three times across two different printers (Ender 2, CR-10), out of 20 or more print jobs I've sent with 3.6.

**Expected results**
The printer should stop printing when the print job ends.

**Additional Information**
Octoprint 1.3.9.

cityeyes, 2018-11-17:
I've been having the same issue. I initially thought it was some sort of user error, but it's only happened with the newest version of Cura and now I'm a bit relieved to see that other people are having the same problem.

I don't have Cura logs, but I may be able to grab Octoprint logs. I **do** have a timelapse of the print, so at least you can see the physical aspects of the issue. 

You can see it [here](https://drive.google.com/open?id=1ciwOk5eUW0VxE1ElxwOsncEp49LvDg2b).

Long story short, the print finishes, sends the printer to ready state, waits a minute or two, then starts the second print, colliding with whatever has already been printed.

closed 2018-11-18 by Appesteijn

eroboticdude, 2018-11-29:
I also have been having this issue. the print head collides with the previous print and leaves scars, ruining the print. It has happened to me multiple times. Where can I find the further discussion?

eroboticdude, 2018-11-30:
> Just upgrade the Cura Octoprint plugin to v3.5.3, it's fixed now.

Where can I find it?

jarrodasmith, 2018-11-30:
Under the "Marketplace" menu in Cura.

jarrodasmith, 2018-11-30:
Under the "Marketplace" menu in Cura 3.6.x

fieldOfView, 2018-11-30:
The OctoPrint Connection plugin is not maintained by Ultimaker. It has its own issuequeue here:

https://github.com/fieldofview/OctoPrintPlugin/issues

Specifically this issue was moved to https://github.com/fieldOfView/OctoPrintPlugin/issues/94 , and fixed in OctoPrintPlugin 3.5.3

chuckdaniel, 2018-12-16:
Not so sure this is an issue with the plug-in.    I have been having this occur in 3.6 specifically when nothing should be even sent to the plug in.    To recreate: 1) disconnect the printer in Octoprint, 2) click "Send to OctoPrint" in Cura - you will get an error message and an option to Queue the job.   Do not do that.   Instead close the message by hitting the X.  3) then reconnect the printer in Octoprint and click "Send to OctoPrint" again.     

You will end up with 2 print jobs merged together resulting in a very bad situation for the printer.    I've also noticed that if you accidentally end up with that queue message and close and reopen Cura, the double print doesn't happen.   This behavior only occurs in 3.6 for me, not 3.4 and back.

jarrodasmith, 2018-12-16:
It’s the plugin. Fixed in v 3.5.3 of the plugin.