Ultimaker/Cura #4826: Material settings - Cannot read anything in darkmode

OliverWieland, 2018-11-18:
**Application Version**

3.6.0

**Platform**

Ubuntu 18.10

**Printer**

Anet A8

**Steps to Reproduce**

- Change color theme to "Ultimater Dark"
- Open Material settings

**Actual Results**

The color font color is nearly the same than the background color

**Expected results**

The background color should be dark, as in all other dialogs

**Additional Information**

[image]

alekseisasin, 2018-11-19:
It is very strange. I checked on Win10 and Ubuntu 18.10, and I have proper dark theme style in Materials settings.
I do not know what to suggest to do. I can say it works well on my computer.

diegopradogesto, 2018-11-20:
This may happen, because this dialog uses the system palette instead of our internal dark theme. This is an example how it looks like in my Windows laptop:
[image]

That being said, if system palette indicates to use black background and we use black text (or the other way around), this will happen.

diegopradogesto, 2018-11-21:
Developers, look at internal ticket CURA-5955

yodor, 2019-01-09:
I was able to partially mitigate this using environment variable QT_QUICK_CONTROLS_STYLE=Desktop
After using this variable my menus and other elements are rendered mostly correctly but still the materials List/View is using incorrect palette.
[image]

Ghostkeeper, 2019-02-05:
Yes, the materials list there was custom made without the palette since the system-styled elements were not powerful enough for what we needed it to do.

bliser3d, 2019-03-19:
Thanks **yodor** for the prompt.
Used the command "QT_QUICK_CONTROLS_STYLE=Desktop /home/User/Cura/AppRun".
It helped me.

LightTemplar, 2019-12-25:
Same here. Using the last AppImage in KDE Neon, dark theme. 
[image]
And with "QT_QUICK_CONTROLS_STYLE=Desktop" it looks not much better, even main menu have become unreadable: 
[image]

Ghostkeeper, 2020-07-06:
We have some duplicates going on here. Please continue the conversation here: https://github.com/Ultimaker/Cura/issues/5490

closed 2020-07-11 by Ghostkeeper