Ultimaker/Cura #4827: úressimg nutton save to removable drive does nothing in version 3.6

jarabizna, 2018-11-18:
**Application Version** 3.6

**Platform** Windows 10

**Steps to Reproduce** Save button changed to Save to removable drive after smartcard insertion into NB. But pressing button Save to removable drive does nothing. It is possible to save to removable drive using general Save option.

**Actual Results** .gcode file is not saved on removable drive after pressing Save to removable drive, option to Remove drive is not displayed.

**Expected results** .gcode file is saved on removable drive and option to remove drive is displayed.

**Additional Information** Test with removable drive containg file with and without same name.

alekseisasin, 2018-11-19:
Does it save on primary drive? Did you try another removable drive?
Can you add the log file?

jarabizna, 2018-11-19:
.gcode file can be saved to any drive (including removable drive identified by cura) using Save to File option (in version 3.5.1 is Save to removable drive working correctly). 

Pressing Save to removable drive generates following error in log:
2018-11-19 10:50:15,622 - ERROR - [MainThread] UM.Logger.logException [84]: Exception: Unable to write to file 3S_podpera: 'application/x-stl-ascii'
2018-11-19 10:50:15,624 - ERROR - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
2018-11-19 10:50:15,625 - ERROR - [MainThread] UM.Logger.logException [88]:   File "X:\3.6\build\inst\lib\python3.5\site-packages\UM\Qt\Bindings\OutputDeviceManagerProxy.py", line 126, in _writeToDevice
2018-11-19 10:50:15,627 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Program Files\Ultimaker Cura\plugins\RemovableDriveOutputDevice\RemovableDriveOutputDevice.py", line 60, in requestWrite
2018-11-19 10:50:15,628 - ERROR - [MainThread] UM.Logger.logException [88]:     file_formats = [format_by_mimetype[mimetype] for mimetype in machine_file_formats] #Keep them ordered according to the preference in machine_file_formats.
2018-11-19 10:50:15,630 - ERROR - [MainThread] UM.Logger.logException [88]:   File "C:\Program Files\Ultimaker Cura\plugins\RemovableDriveOutputDevice\RemovableDriveOutputDevice.py", line 60, in <listcomp>
2018-11-19 10:50:15,631 - ERROR - [MainThread] UM.Logger.logException [88]:     file_formats = [format_by_mimetype[mimetype] for mimetype in machine_file_formats] #Keep them ordered according to the preference in machine_file_formats.
2018-11-19 10:50:15,633 - ERROR - [MainThread] UM.Logger.logException [88]: KeyError: 'application/x-stl-ascii'

Inserting/Removal of sdcard, usb flash while CURA is erunning generates no message in log.

jesseab, 2018-11-23:
Same problem here, trying to save a simple cube to print on a Makerbot Replicator

owendelong, 2020-09-24:
I'm now seeing this same issue in 4.7.1.
The following changes occurred between when I was having success and now:
1. Added a Lotmaxx Shark printer (non-network)
2. Configured Lotmaxx printer for second extruder
3. Added G29 to Begin G-Code to enable auto-leveling for Lotmaxx printer

Am slicing a small file for test print and trying to save to removable.

owendelong, 2020-09-24:
Switching to Save to File and choosing the same removable drive allows me to save without issue.

Ghostkeeper, 2020-09-28:
Indeed, this reproduces. After some investigation I've made a fix here: https://github.com/Ultimaker/Cura/pull/8459

commit a747ca06 (2020-10-30): Fix filtering of output MIME types removable drive

The RemovableDriveOutputDevice was selecting the preferred format to save the slice output in automatically. To select this, it was taking the intersection between the available output formats (for which there is a Writer plug-in) and the file formats that the printer supports. However if the printer supports a file format that is not supported by Cura, it would crash here because it was looking up the plug-in for a MIME type it doesn't know about. Stupid mistake. But not half as stupid as the bug I'm about to fix in the next commit.

Fixes # [... cut]
closed 2020-10-30 by konskarm