Ultimaker/Cura #4828: Increase size of the settings area

dmitrygorbenko, 2018-11-18:
Hi,

the settings area is too small compared to the other sections of the right part of the panel.
Check the image below and you'll understand what do I mean:

[image]

Can you make it bigger ?

Thanks,
Dima

fieldOfView, 2018-11-18:
There's the so aptly named "Prepare Stage Compactor" plugin that does just that. Thing is, the version for Cura 3.6 needs testing. If you are interested in a version of this plugin for Cura 3.6, please help test this prerelease:

http://files.fieldofview.com/cura/PrepareStageCompactor-v5-2018-10-29T14_11_04Z.curapackage

Just drop that file into a running Cura window and restart Cura.

dmitrygorbenko, 2018-11-19:
YEAH !! Much better ))
Thank you !

Would be great to have this plugin somewhere around for those who want to enable it thr the Cura Settings, I think.

Thanks

nallath, 2018-11-20:
The UI will get a redesign in the next release.

closed 2020-11-26 by nallath