Ultimaker/Cura #4829: Support brim AFTER skirt

mbmosher, 2018-11-18:
**Application Version**

3.6

**Platform**

OSX

**Printer**

Ender3

**Steps to Reproduce**

Use the new 'enable support brim' with build platform adhesion type > skirt (tried 3 line and 4 line skirts)

**Actual Results**

It actually prints the support brim first, then the skirt.  So far in all my prints with PETG the material isn't fully flowing when it starts the support brim and several rounds are missed.  Usually it gets going during the skirt printing.

**Expected results**

I would expect the printer to print the skirt first, then the support brim.  This would also allow for checking placement of modle on the bed via the skirt before the brim starts.

**Additional Information**

lsidor, 2018-12-07:
I have also encountered this error.
The skirt should definitely be printed first since the purpose of it is to prime the extruder.
The support brim on my print was under-extruded at the beginning since the extruder hadn't been primed yet.

Derwodaplastik, 2018-12-07:
Same here. I had to print without the support brim because of that. But it is a great feature so please change this. Beside of that Cura is a great software.

MrStump, 2018-12-12:
Same here, please, fix this.

justinianpopa, 2018-12-13:
+1 for adding an option to control this; My printer doesn't print critical small support adhesion brims and it's ruining the whole print. Temporary fix is to add a "prime-nozzle-in-a-long-line" gcode at the start of the print

justinianpopa, 2019-01-03:
any news on this from the dev team? is it just normal behavior or a bug?

cds333, 2019-01-03:
Yes please fix asap. Supports have trouble adhering at the bast of times, this pretty much ensures they will fail.

cds333, 2019-01-03:
Yes, please fix asap! Supports have trouble adhering at the best of times, this pretty much ensures they will fail. And skirts whole purpose is to prime the nozzle while not printing anything that will be part of the final print or even have a second layer, so no- it is definitely not normal behaviour. Shirts should always print first.

paukstelis, 2019-01-11:
Additionally, at least in 4.0.0 alpha, if Build Plate Adhesion is None, support brim is ignored, even if the box is checked. There would be plenty of cases where you want the support brim but not other types of adhesion.

justinianpopa, 2019-01-15:
@paukstelis That seems pretty useful, you should consider opening a separate issue for this

muenznej, 2019-01-16:
seriously this support brim and skirt print order is a mess...  i want the supports to be properly adhereing not the skirt

marekruzicka, 2019-01-17:
Have the same issue... 
We just need to have the priority set properly:  skirt > any raft/brim

In the mean time, use support floor... not as good as brim, but better than nothing.

danspamman, 2019-01-30:
I would like to request that the order of support brim and skirt get switched as well.  Printing the support brim first makes for a nasty mess - the skirt needs to go first.  I'm assuming this is an accidental bug, rather than a feature ?

Kzar42, 2019-02-11:
I'm having the same problem with the support/support brim being printed before the skirt. With small support features this renders the support brim a problem rather than a solution - the extruder hasn't yet been primed, the support brim isn't printed or doesn't attach, then on the next layer the material prints in the air and is carried and deposited elsewhere on the print... It's all downhill from there.

KyleBorth, 2019-02-13:
Would also like to request this to be fixed ASAP. Printing the support brim first basically causes it to not print at all as the extruder has not had time to prime. Thanks in advance!

Jaseduar10, 2019-02-18:
same here, please fix

rburema, 2019-02-22:
Devs: Ticket is in the board as CURA-6002

xkrull, 2019-03-23:
Version **4.0.0** - nothing has changed!
First border (if, of course, it will work out, especially on PETG), and then skirt too :(

rburema, 2019-03-26:
@xkrull,
Sadly, CURA-6002 isn't done yet. I hope we can start on it this week. CURA-6124 was fixed, but only for the master branch, which means it's a lock-in for 4.1.

Schleichmichl, 2019-04-04:
same here, please fix

krz4tw, 2019-04-04:
Exact same Problem please fix asap.

krz4tw, 2019-04-04:
Also a connected issue:
The support brim direction. What I mean by that is that in some cases it starts to print the support brim "islands" on the outer most line and then spiraling inwards and sometimes it does the opposite, start inside and spiral to the outside.
Starting inside and spiraling out is bad because of the short lines with sudden changes of direction make it really hard to stick.
Better to start outside have one longer continious line with smoother curves and then spiraling inwards. The inner sections adhere much better this way cause they can not only stick to the buildplate but also to the lines next to it.

In any case, please fix the oder of skirt and brim first before you consider this.
Thanks

Ghostkeeper, 2019-04-05:
> The support brim direction. What I mean by that is that in some cases it starts to print the support brim "islands" on the outer most line and then spiraling inwards and sometimes it does the opposite, start inside and spiral to the outside.

It's impossible to make this work for any shape. Take this one for instance:

[image]

If it's drawing a brim from outside to inside (=far from the shape towards the edge of the shape) then inside the two "lagoons" in that shape it must be drawing the brim from inside to outside (again starting far from the shape, going towards the shape).

smartavionics, 2019-04-07:
Maybe fixed by https://github.com/Ultimaker/CuraEngine/pull/1029.

Ghostkeeper, 2019-05-29:
Fixed in 4.1. Thanks!

eugr, 2020-02-26:
Looks like this is reintroduced in 4.5-beta :(

Ghostkeeper, 2020-02-28:
I'm unable to reproduce that. Here's a print where it printed the skirt first, then the support brim:

[image]

Do you have a project file that reproduces this bug?

eugr, 2020-02-28:
@Ghostkeeper - the bug appears only when using tree supports. I'm attaching a picture and will try to attach a project file next. As you can see it started to print the support brim for half of the supports first and then the skirt.

[image]

eugr, 2020-02-28:
Here is the project file in 3MF format. I'm using my own printer profile but that shouldn't matter - exactly the same thing happens when using built-in Prusa MK3S profile too.

Notable settings:

- Adhesion type: Skirt
- Tree supports enabled
- Regular supports disabled
- Support brim enabled with default parameters

[support_brim_bug_example.3mf.zip](https://github.com/Ultimaker/Cura/files/4268382/support_brim_bug_example.3mf.zip)

Ghostkeeper, 2020-03-04:
Indeed with that model it reproduces. In fact, not only with tree support but also with ordinary support.

Vandrasc, 2020-03-06:
Devs, for our internal reference CURA-7284

rburema, 2020-03-17:
@Ghostkeeper Not seeing this with ordinary support, only with tree support. If I turn off tree support and enable normal support, skirt comes first. I've flipped back and forth a few times now to check. Do you have a project file where this happens wit ordinary support as well?

rburema, 2020-03-17:
Ok never mind, I see the skirt happens _in between_ the parts of the tree support for me, which means it's probably not as predictable as all that (which makes me doubt very much if my last comment is always true).

gaara92, 2020-04-02:
This bug also affects me, on Cura 4.5 (AppImage@Ubuntu).
If I deactivate the support border (french translated), the bug disappear.
[image]

rburema, 2020-04-02:
@gaara92 It [took](https://github.com/Ultimaker/CuraEngine/pull/1218) [a few](https://github.com/Ultimaker/CuraEngine/pull/1221) [tries](https://github.com/Ultimaker/CuraEngine/pull/1228), but it's considered fixed for the upcoming 4.6 release.

closed 2020-04-02 by rburema