Ultimaker/Cura #4830: [Suggestion]  Summery list of crucial/critical setting's values

Thisismydigitalself, 2018-11-18:
It be nice to have the ability to checkmark selected settings for these to be visible for the user at all times on main slicing workspace. before longs prints i spend time opening all right pane settings to make sure I haven't messed up any crucial settings. it will be nice to have a dynamic summery like:

Layer: 0.2
Walls: 2
infill: Gyroid@10%
Btemp: 80C
Nozzle:210C
Material: PLAx
Support: ON
etc..

this list will have only what i selected to be visible at all times. this can also be included as metadata to profile names.

A penny for your thoughts.

diegopradogesto, 2018-11-20:
I agree and maybe it's a nice idea to have this as a plugin instead of in the main code base.

GregValiant, 2024-11-24:
FieldOfViews "SideBarGUI" plugin provides some of this functionality.
I'm cleaning house here.  It's been 6 years so can this be closed?

closed 2024-12-09 by github-actions[bot]