Ultimaker/Cura #4831: 3.6.0

danstancil, 2018-11-18:
**Application Version**
3.6.0

**Platform**
MacOS High Sierra

**Printer**
printrbot simple metal

**Steps to Reproduce**
Attempt to print any file

**Actual Results**
The software indicates the printer is connected, but the printer does not respond to any command, either printing, heating, or position jogs.

**Expected results**
Printer should properly respond to commands

**Additional Information**
When I revert to 3.5.1, everything works ok again.

nallath, 2018-11-19:
Could you please share your logs?

ksoviero, 2018-11-25:
@nallath I have the same problem on Linux Mint 18.3 (Ubuntu 16.04), and I also have a Printrbot Simple Metal. That's probably not a coincidence. 

I've attached my Cura log as requested. The log contains me starting Cura, loading an object, and attempting to print to USB. Upon doing so the printer does nothing and the monitoring page is "empty". I'll also attach a screenshot of what I mean.

Oh, and also, I loaded up the version 3.5.1 and the printer still works there, so it's a software bug in 3.6.

Thanks

[image]

[cura.log.gz](https://github.com/Ultimaker/Cura/files/2612887/cura.log.gz)

danstancil, 2018-11-26:
I apologize for the delay; I believe I have attached the appropriate log.

Thanks,
Dan

nallath, 2018-11-26:
We don't have a printerbot here, so it's really not something we can test for. All I can see from the logs is that it does seem to be able to detect & connect to the printer.

danstancil, 2018-11-26:
Thanks for checking. There is a simple solution for me: keep using the
previous version of Cura!
Note that a subsequent post from Kevin Soviero indicated the same problem
with Linux and a Printrbot Simple Metal.

InnovationSystemsAU, 2018-11-28:
Geday,

I have the same issue here with a printrbot PLUS.

Logs attached.

Hopefully someone can find the cause as I'd love to use the new features.

[cura.log](https://github.com/Ultimaker/Cura/files/2623065/cura.log)

closed 2020-06-20 by no-response[bot]