Ultimaker/Cura #4833: Valid STL reported as invalid.

inakiam, 2018-11-19:
**Application Version**

3.6.0, 64 bit
**Platform**

Windows 10, 64bit
**Printer**

Anycubic Mega I3
**Steps to Reproduce**

Download this: https://www.thingiverse.com/download:5629654

Extract the mesh.

Attempt to open the mesh.
**Actual Results**

 Cura will report the mesh is invalid. Other slicers open the mesh. The mesh was exported from Fusion360 from a solid made in that program.

**Expected results**

Cura should open the mesh.

**Additional Information**

I've renamed the file to a.stl and attempted to open it to determine if it was an issue with the filename. The renamed file also failed to open.

smartavionics, 2018-11-19:
FreeCAD says it has self intersections - does this matter?

[image]

Thisismydigitalself, 2018-11-19:
Sliced with Cura 3.6.
What am i missing?..
[image]

nallath, 2018-11-22:
I'm not able to reproduce this. The model loads without issue for me. Could you share your log files?

nallath, 2019-01-07:
Closing this due to inactivity.

closed 2019-01-07 by nallath