Ultimaker/Cura #4835: Need a way to choose order of objects for Print sequence - One at a Time printing [Cura 3.6]

printingotb, 2018-11-19:
**Application Version**
Cura 3.5 and 3.6

**Platform**
Win7

**Printer**
Wanhao Di3

**Steps to Reproduce**
Put an object in each corner on the build platform
Set Print sequence to "One at a time"
Slice and take note of the order they are printed in
Switch positions of the objects and re-slice

**Actual Results**
The objects will always slice in an order based on their position.
(In my case Back right, front right, back left, front left)

**Expected results**
Older Cura versions allowed us to choose order by importing the objects in a given order. (First object printed last). Their placement did not affect the print order.
I need a way to be able to choose print order.
The thing is that my hotend setup takes up more space behind the nozzle than at the sides and front. If I could choose print order, I could place a row of objects at the front, a row in the middle and a row at the back and print them all one at a time.
As it is now, I can only print two or maybe three small objects, since it prints in the wrong order for my setup. 

**Additional Information**
Nr 1 will collide with my cooling solution when it starts printing nr 2
[image]
In older Cura, I could re-arrange the objects to change print order
[image]

nallath, 2018-11-19:
If your setup takes more room than that's what should be set in the machine definition. That way Cura can take into account in what order (and how close) objects can be printed. 

As for objects printing in a certain order; That is the actual intended behavior. Loading the objects in a certain order should not interfere with how (and in what order) objects are printed.

printingotb, 2018-11-19:
Sorry if I'm explaining this poorly.
Lets say we have a printhead that takes up 100mm behind the nozzle and only 10mm in the front and sides. 
Printbed is 200x200mm.
Import two 10x10x10mm cubes.
If we start printing at the front of the printer and continues backwards, we should only need 10mm of clearance between each cube in the front/back-direction.
If we print the same cubes starting from the back, we need 100mm clearance between the cubes.
When I enter Ymin 100mm in the printhead settings and try this, I get an "unable to slice" message if the grey area overlaps with another cube even though it would be possible to print this.
[image]
[image]

MiKelloggs, 2018-11-20:
I am having the exact same issue. I think the specific change was in 3.5, as shown here.

[image]

I am in the same boat. I have a large build plate. 300x300. So I like to print a few objects at a time. And I usually orient them and load them in so that my first object printed is at the front of the build plate, then the next object to be printed is slightly further back, going until the last item printed is at the back of the build plate. Printing this way allows me to print more objects without worrying about the X carriage of the printer hitting parts. If you print back to front, it will hit.

Here is a post I made on the CR10 facebook page explaining it a little more in depth.

"Did Cura change the one-at-a-time printing order? I used to be able to load what I want printed last, first. And as I load objects in they are printed prior to what has been loaded in, meaning the last thing I load in, gets printed first, second to last is second. Now every time I try to move objects around and slice the print order changes. I like to order my prints and set them apart on the build plate printing front to back, and flipping from the right side to the left so my X carriage or hot end doesn't knock them off... this is really screwing me over right now.

In this picture I want to print the one closest to the origin first, then the one above that and all the way to the right, then print as they move further back in the build plate. It prints the one by the origin, then the top left, moving towards the front, and the lastly the one in the back corner. If I try to move them around so that the order would be right, it resets!!!"

[image]

nallath, 2018-11-20:
@printingotb; The second image should result in something that can be sliced. I've just tried it with master and it correctly sets the order (instead of complaining about that it doesn't fit).

So as long as your print isn't higher than your gantry, it should do it correctly. The previous behavior was actually a bug; it would completely ignore the head size and only allow for manual selection (by means of a super weird UI flow of "What order did you load them in").

MiKelloggs, 2018-11-20:
@nallath, Whether or not the load in order resulted in the print order being a bug doesn't really matter. The fact is that that 'bug' gave us a way to control what order objects were printed, and now there is no way to control that. On the topic of gantry height, most people enter a very high number so the slicer doesn't get mad when you print tall objects. The gantry height doesn't come into play if we are able to set a specific order, and start printing from the front of the build plate to the back, and the gantry never crosses the path of a printed part. 

I know that some of this is to help people avoid failed prints from them being knocked over but for the more experienced printers, there needs to be a way to control all of this.

nallath, 2018-11-20:
I've recently fixed an issue with the head sizes in master, so at least those sizes should be correct for the next release. 

As for reverting the changes; we're not going to do that since it would introduce issues that it sought to fix. It's unfortunate that this broke things for you, but it was never intended to work like that. The print one at a time behavior is now as it should be for Ultimaker printers (The changes as you want them won't have any advantages for Ultimakers). This, unfortunately, means that I won't spend the time to add the "set print order" feature. 

So if you guys want it back, I'm going to have to fall back to the old "Pull requests welcome"

MiKelloggs, 2018-11-20:
@nallath that's a big bummer. Even just a simple toggle or settings option deep in the menus, that defaults to the movement as of this update, but can be turned off would be nice. 

It sucks because I love Cura and all of the new updates like gyroid infill are cool but I really need this functionality. I can't really afford $150 for simplify3d but I might just have to now :(

nallath, 2018-11-21:
If only it were as easy as just adding a single option, we wouldn't be having this discussion (I would have just added it in that case).

MiKelloggs, 2018-11-21:
@nallath I totally understand. Thanks for the response is so that we know what's going on!

nallath, 2018-11-22:
It's actually a change in Cura Engine (https://github.com/Ultimaker/CuraEngine/pull/802)  based on a  bug report (https://github.com/Ultimaker/Cura/issues/3296). 

Actually adding a selectable order (all the while ensuring that the original bug doesn't pop back up) will be quite difficult I'm afraid.

rlabusiness, 2019-02-22:
> It's actually a change in Cura Engine ([Ultimaker/CuraEngine#802](https://github.com/Ultimaker/CuraEngine/pull/802)) based on a bug report (#3296).
> 
> Actually adding a selectable order (all the while ensuring that the original bug doesn't pop back up) will be quite difficult I'm afraid.

Ok... So Ultimaker developers shy away from solving difficult problems, even when the value of such solutions is very clearly stated by a massive community... gotcha. Good to know.

Add me to the list of this being needed. It sort of blows my mind that this wasn't obvious.

nallath, 2019-02-25:
You are partially correct. Since our primary responsibility is towards Ultimaker customers, that is where our focus is. The problem just has a lower priority because it doesn't impact Ultimakers, it impacts third-party machines. We never made a secret out of that, so there is no need to make any snarky posts about it. 

That being said, we're more than willing to give any advice to people who are willing to fix it, we just won't do the actual work.

Ghostkeeper, 2019-02-28:
The primary problem with letting users select the order of printing one-at-a-time, is that the collision areas of the models are going to change based on the order that you're printing in. In printingotb's comment [here](https://github.com/Ultimaker/Cura/issues/4835#issuecomment-439959685) you can see that effect: Depending on the printing order, the objects either collide or they don't. If the collision areas are all over the place that is going to be very confusing for the user, especially because some objects are allowed to be in the collision zone but some are not.

The intent is currently to take the shape of the print head, rotate it 180 degrees around the nozzle position and take the intersection of that and the original print head shape, then take the Minkowski sum of the result with the model's convex hull. Then we calculate the order in which things can be printed (which is then always possible unless there is a limitation in the Z direction).

seltix5, 2019-06-19:
+1 on the manual order selection... it wold be super usefull to maximize the print area :D

mjy90, 2019-06-30:
+1 for manual order selection. Specifically, I'd like for the order to take into account the heights of my parts. If only one part is taller than the gantry clearance, it could safely be printed last. However, the current "all or nothing" logic says I can't do the following:

[image]

krishjun, 2019-07-12:
I would like to add following suggestion 
1) Skirt can be 1 line, but in squre shape automatically touching the boundary by default of build plate.
2) With no skirt shadow will be smaller .

dekerser, 2019-07-22:
I ran in to the same issue. Decided to create a script for it. It is just a first go at it and probably needs more work. It also could be a gcode script, but for now this is just a standalone python script. It does have a few assumptions for the gcode, but feel free to adapt. 

Usage: change_order.py <gcode file> 9 6 3 2 5 8 7 4 1
The numbers are the order of placement by Cura. So the 9th placed object is now becoming the first object. The 6th placed object is becoming the second.  See attached images (for clarification) 

Attached two images and the python script (perhaps it needs to be somewhere else in "GitHub", please feel free to do so and leave a reference here).
Images attached, rgb marked corner is front left:
change_order.png: new print order from front to back and left tot right and right to left for the middle row. 
cura_order.png: begins at the back row working its way to the front, and repeats that 3 times from right to left.
[image]
[image]
[change_order.zip](https://github.com/Ultimaker/Cura/files/3419387/change_order.zip)

Ghostkeeper, 2019-07-23:
When using that, keep in mind that the disallowed areas (the shadows around the models) are constructed such that it works with the order that Cura thinks of. If you change the order, the print head can collide with the previously printed models if it is not symmetrical around the nozzle.

dekerser, 2019-07-24:
Indeed, but that is exactly the reason for changing the order. Which means: the Cura order is wrong for my printer, a renkforce RF100v2, and makes the print head's X-mover crash into my previous printed object. The head's X (red axis) - mover is at about 1 cm above the plate, or more accurate, 1cm higher than the nozzle tip,  so when moving in the Y direction from back to front it will kick over or block on any print higher than 1 cm.  Perhaps there is a setting for it, but I couldn't find it. Also the X-mover
has 3 cm width, so moving from back to front will never work efficiently. Hence the script.

mjy90, 2019-07-24:
@dekerser, indeed there is a setting for that! If you go to Settings -> Printer -> Manage Printers -> [select your printer] -> Machine Settings, you'll find it under Printhead Settings, as below:
[image]

Ghostkeeper, 2019-07-25:
> the Cura order is wrong for my printer, a renkforce RF100v2, and makes the print head's X-mover crash into my previous printed object.

The Cura-devised print order is meant to guarantee preventing crashes between the print and the print head. This is important to us because crashing print heads with the print can damage the printer. Of course, this guarantee is only valid if the print head size is correctly filled in. For your Renkforce the print head size seems to be wrong.

I'm afraid that introducing a setting to change the print order can
* allow the user to accidentally choose an order that causes the print head to crash,
* only reduce the space available to your print, since the order that Cura chooses is optimal to minimise space usage.
* be just as hard or harder to discover than the printhead settings in Machine Settings
* not be set correctly in the definition files.

So in those four ways, the current solution is better. Of course, the disadvantage is that it requires the print head to be correctly set, which is a problem if you have a custom FFF printer, but then we show the dialogue in your face when you add the printer anyway.

seltix5, 2019-07-25:
hello @Ghostkeeper ,
about your points, I dont think thats true...
- allowing the user to choose the order will not be worse than let Cura chose the orders without allowing the user to change it... see this examples :
[image]
[image]
even if that is a problem, there is other important settings the user could mess up, just put the option in the export mode or something like that....
- I dont understand why the auto order by Cura chooses the optimal to minimize space usage since it will not move the objects

About the printhead settings, if this settings are the problem, how do we measure the X/Y min/max? I have a ender 3 with the default printhead settings and if I leave the default gantry height I cant even print bigger objects... witch does not make sense since if I print in the Y direction the gantry height is not a problem...

Ghostkeeper, 2019-07-26:
I'm not really following your first point. What's wrong with either ordering?

About the second point though, it's a bit hard to explain, but let me try...

Let's say you have a printer with a print head that is a giant 10cm cube, but the nozzle is located in the front left corner. You need to print lots of objects so you want to fill the build plate as much as you can.

If you start printing an object from the front left corner of your build plate, the second object can be placed directly adjacent to the first. This is fine, because the 10cm cube of the print head is located on the other side of the nozzle, to the back and to the right of your nozzle. You could put the second print either on the right of the first, or more towards the back. Either way, your print head will not collide with the first object. Then the third object can again be placed to the right or to the back of the second, as long as you're consistent.

If you'd start printing from the back right corner of your build plate, the first object still prints fine. But to print the second object, you'd need to print it >10cm to the left of the first object, because there is a giant print head in the way. Or you'd need to print it >10cm in front of the first object, because there is that 10cm block on your print head.

Cura encodes the "shape of the print head" in the shadow area around your objects, if you're printing in one-at-a-time mode. It does an operation on the shape of the print head that reduces the area (polygon intersection with its 180 degree rotation) and then devises the order in which it must print in order to be able to print so close together.

If the order changes, so must the disallowed areas. The shadows around the object can only grow bigger, because the order that Cura devises is optimal to minimise the size of the shadows.

> About the printhead settings, if this settings are the problem, how do we measure the X/Y min/max?

The X Min is the position of the left hand side of the print head, relative to the nozzle position. The X Max is the position of the right hand side of the print head, relative to the nozzle position. So the X Min is normally negative and the X Max is normally positive. Similarly, the Y Min is the position of the front side of the print head relative to the nozzle position and the Y Max is the position of the back side.

> I have a ender 3 with the default printhead settings and if I leave the default gantry height I cant even print bigger objects... witch does not make sense since if I print in the Y direction the gantry height is not a problem...

I understand now that you're talking about maximising the height of the print volume, while I was talking about maximising the usable build plate area. In theory the last object can be printed higher than the gantry height because the nozzle doesn't go lower than the gantry any more after printing it. Cura doesn't take this into account except that it allows the full build volume if you have only one object to print. That is a flaw that can be improved upon, but that'd mean that the height of the build volume would change depending on which objects you loaded and how you rotate and place them, which could be a bit weird for the user to see.

seltix5, 2019-07-26:
hello,
yes, your'e right, I think this is all related with the gantry height, and from what I read online, I think probably the majority of the people just want the manual order because of that.
From every post online the first thing you find you can to is increasing the gantry height to be able to print sequential and bigger than 3cm (in my case). After that you almost forget about that setting and the problem became the print order, because to print bigger objects they need to be printed in the correct order to prevent crash or for example, in my case, sometimes I print half of the build because I will start a second print remotely without removing the first print, if I cant print in the correct order, independently of the objects height, the print head may finish and the wrong place for my second print depending on my START/END gcode ( if I have my build plate moving forward at the end of a print for easy removal I may not want to finish printing smaller objects at the front, it will crash when moving the plate, unless Cura move the z axis to the bigger object height after finish? )
If you check all the previous printscreens does not have a gantry height because otherwise sequential print wold only work to print very small object and wold not be as useful as it could be.
About the possible implementation I dont think it wold be weird to see, a simple implementation wold be : the build volume is exactly the same when printing all at the same time, but when printing sequential, instead of reducing the build volume to the gantry height, that height limit wold just be a second "line" on the build volume to inform the gantry height.
We are all asking for manual sorting because it is the easiest and fastest way to maybe get it and because each one of us has one or another reason to print in that specific order, and auto order will never fully satisfy us all for X reasons.
And about the possible problem to users, advanced users know what they are doing (this is not my case, i'm still learning) but I'm informed enough to know in what order it will not crash for my specific printer. Just adding the option disabled by default with the option to enable it in the settings options with a big alert to warn users when enabling it wold be great and the freedom we want in this cases :)

> The X Min is the position of the left hand side of the print head, relative to the nozzle position. The X Max is the position of the right hand side of the print head, relative to the nozzle [...]
ok thanks! but, the X min is normally negative? the Y min too? the default printhead settings for ender 3 is Xmin 20 / Ymin 10 / Xmax 10 / Ymax 10, there is no negative value.

Thank you

LD3D, 2019-07-29:
Hi sorry but I am a bit confused with all the back and forth on this topic.
Here is my issue and is there a fix for it....?

IN Cura 4.1.0 using ONED AT A TIME order... if I place many models on the build bed and slice Cura always wants to start printing from the back to the front the left and right don't matter as much.

I have a CR10S Pro it is a gantry style so the optimal print way would be to print models from the Front of the build bed to the back. In this way, the gantry will not smash the parts already printed and let more prints be made on that build bed.

So how do I change the print location to always print the front of the bed to the back of the bed

Could this be changed by resetting the home position on the machine in settings? Or is there a setting in Cura that can change this order of slicing to print order?

Ghostkeeper, 2019-07-30:
You need to set the shape of the print head then to be longer in the back than in the front; further away from the nozzle in the back of the print head (Max Y). This seems to be the reason why you want it to print from front to back anyway.

dekerser, 2019-09-06:
> 
> 
> @dekerser, indeed there is a setting for that! If you go to Settings -> Printer -> Manage Printers -> [select your printer] -> Machine Settings, you'll find it under Printhead Settings, as below:
> [image]

A bit late from my side, but suggested settings were tried, but as said in my previous post, that doesn't change the print order. It just prohibits to print another object as it then within the perimeter of the nozzle.

foulkn, 2019-11-27:
I have been able to change the print order by fudging my print head numbers. 
The actual print head settings of my printer are -
X min -30
Y min -20
X max 30
Y max 60
These settings makes the order go from back right to front left.

I change to these
X min -30
Y min -60
X max 60
Y max 20
now it will print from left to right,  front to back.

So in essence in stead of putting the larger space requirement on the rear of the print head, I put it in the front. My X actually does not matter , but putting a larger requirement on the right of the print head changes to a left to right printing order.
This may help some. It will start with the left objects and work right. Still have not been able to make it start with the front objects and work back.

Cura 4.1
Hypercube.

Noguai, 2019-12-12:
Quote from #6772:
> This was indeed a bug. The order in which models are loaded shouldn't matter (as you say, it's very weird for the workflow). It was causing damage to printers because the print head was allowed to collide with the previously printed models. The disallowed areas are constructed together with an ordering such that this shouldn't be possible, but since the order was arbitrary it was still possible.
>
> The current direction is not necessarily one of the possible configurations that you propose. It could end up printing objects in a spiral or any shape if the objects' positions allow it. The ordering is computed to maximise the available horizontal space. You can't place objects closer together than this because one of the sides of the print head will always hit the other objects. Turns out that even for maximum packing of the build plate with the closest possible distance between all objects, there is always an ordering available in which these objects can be printed. And Cura will automatically find that ordering. There is no possible way in which a user could manually think of a better ordering... Except for one:
>
> The very last object in the ordering can in theory be taller than the gantry height without the gantry colliding with previously printed objects. Cura doesn't take this into account. This is the actual problem you're trying to solve: Cura should allow the last object in the ordering to be taller than the gantry height. However Cura automatically comes up with an ordering so even if the last object is allowed to be taller, it'd be unclear to the user which object that is. As far as I'm concerned, that's where the current problem lies: #6627

> Closing this as a duplicate of #4835. Please continue the discussion there.

@Ghostkeeper: Well then I would argue, that the current ordering, where the gantry is merely regarded as a limitation of the maximum print height, is to be honest not useful (enough). In my opinion the thing that is much more important than print area is print *volume*. Especially when looking at my machine with its mere 25 mm gantry height (in my case the X-Axis), which causes Cura to prevent me from printing anything useful I actually would like to print in sequence.

Also if I lie to Cura about the gantry height and just enter the correct X and Y dimensions of the print head, then manually lay out the models on the print bed (with a nice offset in the Y direction so that neither the X axis nor the extruder and hotend cables would catch a previously printed model, if printed in the correct order), Cura will happily slice the file, not complain about anything, but the order chosen by Cura will physically *not* work. Letting Cura layout the models on the print bed for me automatically does of course not work either.

What I think a better solution would need to do, to work for the use case of most of the people that are not happy with the current ordering algorithm is:
- Just as it does today: Check whether the models on the print bed are far enough apart
[image]

- Additionally for every model *higher* than the gantry height: Regard the orientation of the gantry itself (the direction in which the axes reach over the print bed) as well as the distance between the nozzle and the axes. Meaning for each axis: The min and max coordinates of the bounding boxes of each model in the direction orthogonal to one axis may not overlap more than the distance between nozzle and the axis. So that not just the print head does not collide, but also the axis will not collide.
[image]

- When it comes to print order: Order the models by the coordinate of the center point of the bounding boxes in the direction orthogonal to the axis and also in such a way, so that the axis faces away from the models already printed:
[image]

This should end up producing a zigzag pattern for machines with one axis reaching over the print area and end in a single diagonal line for machines with two axis over the print bed. And the same as today for models smaller than the gantry height.

I know that this sounds like a lot of work to implement and probably not that easy to compute and I also know that I'm pretty much ignoring delta printers as well as some other details, so that the "solution" above is unlikely to be implemented. But with that I want to make the point, that it is actually very sensible to allow the user to manually set the print order when using the "One a a time" print sequence setting. I think also Ultimaker users could benefit from that.

I would also be perfectly fine with that option being hidden until a setting that is buried deep within some menu is activated and Cura displaying a big and fat disclaimer that it cannot check whether my print job will actually work and Ultimaker is not responsible for me damaging my machine by clicking something like "Ignore, continue anyways". There is simply nothing more infuriating when a software is too defensive about checks that are too time consuming or low priority to implement and thus prevents me from using my machine efficiently, just to save novice users from doing something they may not (fully) understand.

**tl;dr**: Letting users manually choose print order is not as bad of an idea as it sounds and actually has some beneficial, valid use cases. So please let us do it, even if it requires activating hidden settings and accepting disclaimers.

Ghostkeeper, 2019-12-13:
If you have a very low gantry, then it's probably more useful to just include the shape of the gantry with the print head and increase the gantry height to the build volume height. That way you'll be able to print tall objects one-at-a-time and it will produce an ordering such as what you propose, making diagonal paths.

Noguai, 2019-12-14:
Thanks a lot for your suggestion, if it works that way, that would be even better. 

Though it seems, that either I am unable to configure it correctly, or the machine_head_with_fans_polygon setting in printer_settings.inst.cfg under definition_changes does not support more than 4 points. As soon as I add a fifth point to the polygon Cura behaves weird (at least I don't understand what's going on). 

I've made two screenshots with the shape I would have expected plotted in GeoGebra side by side to the one in Cura.
[image]
[image]

I inserted a cylinder and made it 0.1 mm small to reduce the effects of the Minkowski sum you have mentioned earlier.

Ghostkeeper, 2019-12-16:
The first screenshot seems logical to me. The shape is rotated 180 degrees and then intersected with the original shape before rotation. That ends up with the shape outlined in red, in your GeoGebra.

The second screenshot makes no sense at all, indeed. You could try reversing the order of the coordinates, since polygons are normally entered counter-clockwise in computer science?

Noguai, 2019-12-16:
Sure, I will do that and post again.

In the meantime I've looked around in the code (though without any Python experience) and found that the call to Polygon.intersectionConvexHulls in ConvexHullDecorator._compute2DConvexHeadMin will only return a polygon with 4 points, if I read the code correctly. I wanted to to change that line and see what happens, but I did not succeed building yet. (I'm new to that black magic that is docker and cmake)

Noguai, 2019-12-16:
With reversed order the shape changes, though still only four points:
[image]

Noguai, 2019-12-16:
Changing that line in Polygon.intersectionConvexHulls did the trick, now the print head shape can have more than 4 points. 

If I now carefully match the head shape to the object alignment on the print head Cura indeed will choose the correct ordering. So **thanks A LOT** so far: 
[image]

Next I just have the problem, that I have to somehow tell Cura not to turn my print head polygon into a convex shape. Uncommenting the calls to getConvexHull in Polygon.intersectionConvexHulls did not work out, yet. Do you have any advice?

Ghostkeeper, 2019-12-16:
Yeah that it only returns up to 4 points looks like a bug to me.

Ghostkeeper, 2019-12-16:
Hmm, I think the intersection operations being made there are only made to work with convex shapes, since it assumes that it's about the shape of the object that needs to be avoided and CuraEngine only guarantees avoidance with the convex shape (and it makes the algorithm a whole lot faster and simpler).

SneakyTiki, 2020-01-03:
I meant to get back to my original post sooner, but life got in in the way (as it usually does).
I intended to make a point about how a computer _should_ be better than me at arranging and optimizing _if_ it had all the necessary information, but Cura knows nothing about the gantry or its location, so it can't possibly generate a useful order..... however looks like Noguai made the point for me. But hey, at least I sort of understand why a manually controlled print sequence will pretty much never be a thing, so guess I'll go make sure my copy of Cura 3.4.1 is backed up so I never lose it.
On the bright side, seems there's a way to define a custom extruder shape at which point Cura will finally do reasonable print sequencing? Sign me up!
Seriously @Noguai I'm not super proficient in code, but if you point me to the correct places and give me some basic instructions I'll muddle my way through. This would be incredibly useful to me, as well as others I'm sure. I looked through the Cura files to see if I could find where you were editing, but the only place I found to reference the machine_head_with_fans_polygon setting was in resources/definitions and the files were .json, whereas your screenshots show an .cfg file. And I failed to find the Polygon.intersectionConvexHulls reference you made which allowed you to add more than 4 points. It'd be nice if the gantry was defined as a separate obstacle to avoid, so the no-no area doesn't have to be a funky diamond shape, but frankly I'll take what I can get. I just need Cura to maintain a reasonable print sequence for my machines, and I'll take it however I can get it. Diamond shapes, round shapes, squiggly shapes, anything. Please teach me your black magic senpai.

Also @Ghostkeeper could you explain why one-at-a-time printing gets disabled if I'm using more than one tool for the job? Is there a limitation of some sort I don't understand? Splicing gcode together is getting tiresome and understanding why will probably make me feel better about having to do it.

Thanks to both in advance

Ghostkeeper, 2020-01-03:
> I looked through the Cura files to see if I could find where you were editing, but the only place I found to reference the machine_head_with_fans_polygon setting was in resources/definitions and the files were .json, whereas your screenshots show an .cfg file.

This modification can be made in either the .json file (I'd advise to copy the setting from a different .json file like UMS5) or in the "definition_changes" CFG file belonging to a certain printer that was already added to Cura. The JSON is meant to be the actual printer definition as Cura loads them. The definition_changes profiles are modifications that the user has made to their printer definition (i.e. the Machine Settings dialogue). I think SneakyTiki made these changes because they can be made without admin access. If it's in both, the CFG file wins.

> And I failed to find the Polygon.intersectionConvexHulls reference you made which allowed you to add more than 4 points.

I've applied the change here.
https://github.com/Ultimaker/Uranium/commit/fedcdfa1520b7212898f83e7343fc79669322276#diff-08bd5f6c53a49ff114c9fa347247b8f9

Looks like I tried to reference this issue but since the change is in Uranium it failed to link it to this issue. You need to modify this in the file `UM/Math/Polygon.py`.

> It'd be nice if the gantry was defined as a separate obstacle to avoid, so the no-no area doesn't have to be a funky diamond shape, but frankly I'll take what I can get.

In the next release (or if you make that modification yourself) it won't be limited to 4 points any more but it will still be convex. The convex shape is taken to avoid travel moves hitting other models.

> Also @Ghostkeeper could you explain why one-at-a-time printing gets disabled if I'm using more than one tool for the job? Is there a limitation of some sort I don't understand? Splicing gcode together is getting tiresome and understanding why will probably make me feel better about having to do it.

We did that because the Ultimaker printers switch extruders by moving the print head to the switching bay. This moves the nozzle outside of the convex hull of the object, possibly hitting previously printed objects. We still intend to solve this. Maybe solving it by moving the build plate down at every extruder switch isn't the nicest solution though. A printer definition can override this behaviour by adding this to the overrides section in the JSON file:

```
"print_sequence":
{
    "enabled": true
}
```

By default it's only enabled if `extruders_enabled_count == 1`.

Noguai, 2020-01-03:
@SneakyTiki I didn't yet have time to further look into that concave-convex-conversion thing, thus you would need to change the print head shape depending on the space between objects you want to print, while not getting maximum efficiency, so manually post processing and merging gcode as you already do might still yield better results regarding that.

The weird shape you see in my screenshot above has been created from:
```
machine_head_with_fans_polygon = [
	[-200, -30],
	[-200, -20],
	[-30, -20],
	[-30, 80],
	[60, 80],
	[60, -20],
	[200, -20],
	[200, -30],
	[10, -30],
	[10, -125],
	[-10, -125],
	[-10, -30]]
```
This is basically the size of my print head where I added a rough estimation for the X axis rods and extruder cables that stick out to the back. Latter is also the reason that Cura will start to print objects from front to back.

As Ghostkeeper said, I have modified the definition changes for my printer profile, as I configured Cura from scratch using the Custom FFF Printer definition, but that doesn't matter. In Windows the file can be found under `%APPDATA%\cura\4.4\definition_changes\`

Also I would suggest to wait for the next release, as for one, the code change I did has already been made by Ghostkeeper and also, I and don't know Python, C++, C-Make, Docker, but I am a .NET developer and just fumbled my way through to setup a build environment and then compile the code, which is rather difficult to do for someone with no development experience at all. The instructions to do so are written down by Ultimaker at https://github.com/Ultimaker/cura-build-environment and https://github.com/Ultimaker/cura-build

Doing it natively in Windows didn't work out for me, probably due to some conflict with an existing Visual Studio installation, but I don't know for sure, I was just getting strange errors while compiling. Building inside a Windows Docker container worked much better for me, but I had to manually download and patch the chocolatey package for `vcredist140` and then modify the file `Dockerfile.vs2015` like [so](https://gist.github.com/Noguai/a602fb19fb4d6872fe399026867a9a6c), as that package was broken for me (vcredist140 did not install due to some broken dependency checks).

SneakyTiki, 2020-01-08:
Well you both are simply wonderful humans, thank you! I didn't have a chance to play around with the information you've provided, but hopefully next weekend.
And to think, instead of stewing in my frustration, all I had to do was ask the right people :D

Magnetous, 2020-03-04:
Hi!
I was struggling to have my parts printed in the order I wanted and I finally found the solution and it is very very simple!
Cura prints the parts not according to their position on the plate but actually according to their position in the parts list at the bottom left of th screen. Cura will print the parts in the oposit order from the list, beginning with the bottom line!
Here is an example:
I have six parts to print, exactly the same. I open the first .stl and then duplicate the part. Cura will then create a list like this:
P
P (1)
P (1)
P (1)
P (1)

Magnetous, 2020-03-04:
Hi!
I was struggling to have my parts printed in the order I wanted and I finally found the solution and it is very very simple!
Cura prints the parts not according to their position on the bed but actually according to their position in the parts list at the bottom left of the screen. Cura will print the parts in the opposite order from the list, beginning with the bottom line!
Here is an example:
I have six parts to print, exactly the same. I open the first .stl and then duplicate the part. Cura will then create a parts list like this:
P
P (1)
P (2)
P (3)
P (4)
P (5)

And no mater how you place the parts on the bed Cura will always print the last part first and then move up the list so you will have:
P(5) > P(4) > P(3) > P(2) > P(1) > P

So if I want the parts to be printed from the front of the bed to the back of the bed I just move them according to the position in the list. P(5) at the very front an P at the back.

If you have to deal with different parts just make sure you open the files in the opposite order of your print sequence. Cura will always put the last opened object at the bottom of the list. So open the parts file beginning with the one you want to be printed last.

I hope it is clear enough, otherwise I can add screenshots to illustrate.

Bye!

printingotb, 2020-03-06:
This is great! It works with Cura 4.5.
4.3 did not work in this way.

Sophist-UK, 2020-04-14:
See #6627 for a discussion on a potential solution.

BenBergman, 2020-04-18:
I am struggling with this issue as well. I am trying to print a lot of small items that are just a bit taller than my gantry gap. I don't necessarily need to specify the exact print order, but I do need to force printing from front to back to avoid gantry collisions.

[image]

Lying to Cura about the size of my printhead will ensure nothing at the back of the plate gets printed before something towards the front of the plate in roughly the same x location, but in the image above Cura tends to try ordering parts in columns, ie. Starting in the bottom left and going up to the top left, then the bottom row second from the left and going up to the top row, and so on.

I've also tried positioning the objects as suggested by @Magnetous but I just get similar behaviour.

The ideal solution would be to check collisions with gantry separately from the print head geometry. This eliminates the current height limit and by checking the two shapes separately eliminates issues of convex hull collisions where there are none.

Barring that, an option to bias the ordering algorithm to start in the front row and move backwards (perhaps with an inverted option for some printers) would solve pretty much all the various issues raised in this and other issues.

As an aside, an additional reason to add this feature is for added compatibility with Octolapse. Most of the time Octolapse users park the print head in the back corner when taking a picture. By printing front to back, Octolapse can be used with multi-object prints and safely avoid collisions.

BenBergman, 2020-04-21:
I've used @dekerser 's python script to reorder the files in the exported g-code as a work around, but as they stated it bypasses Cura's safety checks and requires an extra step after exporting.

I just tried another work around that might work for some, particularly those laying out a grid like my image above. By arranging a single column of items, grouping them, then multiplying the group to make all the rows, then ungrouping, Cura arranged my print order in a row by row order exactly as I needed it. If the order you want is left to right instead of front to back, group the objects in rows instead of columns. I even tried regrouping and then ungrouping after the initial layout and was able to swap the order orientation without having to recreate all the objects.

If that isn't working for you, a half-solution is just grouping objects into rows and leaving them grouped. Anything in a group will be printed at the same time, but at least you are then only printing a couple items at a time instead of (in my case) all 30. This gives the side benefit of letting anything in the same row be closer together as there is no longer any collision risk within the group.

josephted, 2020-05-03:
I also understand that there must be a way for the user to change the order of objects. The program arbitrates the sequence in the order that the objects are inserted on the table. From the last to the first.
There could be a window that allows you to change that order.

fieldOfView, 2020-05-03:
The Object list (lower right corner) could be used for that if it had a way to manually sort the models instead of using alphabetical sort.

Fusseldieb, 2020-05-16:
I also lack an option to set the printing order

Just wanted to print more things at once and am stuck with this problem...

Kravchenko32, 2020-05-17:
Why can't it just at least default the order front to back. I'm trying to print 4 tall identical objects and it's driving me crazy!

muchtall, 2020-06-25:
So partial credit to @BenBergman here, as he had a similar idea. https://github.com/Ultimaker/Cura/issues/4835#issuecomment-617362971

I was able to reliably determine print order of individual models by following this workflow. I hope it helps others out:

### How to specify the print order of Cura's sequential printing feature
- You may want to pause slicing before you start to make the process go faster
- Select the model that you want to print SECOND
- Hold shift and select the model you want to print FIRST
- Group the models (Ctrl+G)
- While leaving the group selected, hold shift and select the model you want to print THIRD
- Group the models (Ctrl+G)
- While leaving the group selected, hold shift and select the model you want to print FOURTH
- Group the models (Ctrl+G)
- Repeat, adding each model to the group in sequence until you have all your models grouped
- Now, with the entire group selected, repeatedly ungroup the grouped models (Ctrl+Shift+G) until all models are ungrouped
- Slice and preview your print sequence to confirm the print order is correct

Sophist-UK, 2020-06-25:
IMO we should NOT provide this functionality (that is assuming that the sequence makes no functional difference to the QUALITY of the prints). IMO the sequence for one-at-a-time printing should be determined by Cura using a simple algorithm which:

1. Ensures (if possible) that the gantry does not hit previously printed objects; and
2. Tries to minimise the print time.

Ideally there should be a button to click which optmises the object locations on the build plate to achieve this in the most optimum way. So in the above animated example, assuming no gantry / head issues, it might put 4 or 5 of these objects on the same row with the skirts almost touching. If the objects were taller and might be hit by the gantry then it would need to position them differently.

muchtall, 2020-06-25:
@Sophist-UK , actually, in my example above, I didn't strictly need the gantry clearance, and I specifically printed one at a time to improve my print quality significantly. PETG is known to be very stringy when doing travels. This leads to strings and zits on the print surface as well as goobers on the nozzle. If you can avoid travels between models, and comb the head over already printed areas without retraction or z-hop, you can almost eliminate those problems.

Sophist-UK, 2020-06-25:
No - you don't need gantry clearance for these small parts, but any solution needs to be generic.

And I do understand why you want to print one-at-a-time - my previous comment was suggesting that Cura should be programmed to automatically optimise one-at-a-time printing as far as possible, rather than relying on the user to specify it manually.

fieldOfView, 2020-06-26:
> Cura should be programmed to automatically optimise one-at-a-time printing as far as possible

Cura also tries to create supports automatically, but still lets the user customize them through support blockers and print-as-support elements. Having the option to customise/affect print order would be nice for advanced users.

Corgano, 2020-11-17:
> So partial credit to @BenBergman here, as he had a similar idea. [#4835 (comment)](https://github.com/Ultimaker/Cura/issues/4835#issuecomment-617362971)
> 
> I was able to reliably determine print order of individual models by following this workflow/workaround. I hope it helps others out:

I really like this idea, but i can't get it to work in cura 4.7.0

I'm trying to print this. My gantry is in FRONT of my x axis - so it is SIGNIFICANTLY easier to print front-to-back. However everything i tried either makes it print all at once or makes it print back-to-front.
[image]
Not giving us full control is one thing, but not even letting us pick the print DIRECTION is another. If i want to print several tall objects in one go, without having ALL of them be scrapped if one fails, it is essential to set print order, and due to my printer geometry optimal to print front to back.
Should I open a new issue for this

Ghostkeeper, 2020-11-19:
No, please keep the discussion together in this ticket. New tickets should be closed as duplicates of this one.

In the most recent few versions of Cura the collision areas of one-at-a-time mode are larger than they need to be. This is a known issue and should be kept open.

Sophist-UK, 2020-11-19:
The issue with one-at-a-time is that it restricts the build volume rather than move items so that the gantry / head will not collide when they are printed.

There may also be an issue of collision areas being bigger than they need to, but that is not the fundamental issue.

Corgano, 2020-11-20:
What I don't get is why this is a problem in the first place.

That list of items in the bottom left corner? Top one prints first. Then
the next. Then the next. Let us click and drag to reorder them. Done. No
new ui elements, it's intuitive, it's simple

If you want to do handholding and stuff then pop up a disclaimer when
someone enables it that says WARNING: THIS COULD BREAK SHIT and link to a
guide on how to use it. For anyone who has used it now than once and gets
the just of how it works the only thing keeping this from being left on
permanently is the lack of defining print order!

My current workaround is
1. Place all the stuff I want to print on the bed
2. Slice it. Check the results in the preview tab
3. Move items into the appropriate positions based on the order they were
last sliced to be considerate of x axis / gantry positioning
4. Slice again and check results. If order changed go back to 3
5. Print.

It can take 5 or 6 tries to get it to print it in the order you want, or to
move the objects to the correct position to match is order. By the time you
get to printing this feature actually works REALLY well. Only thing keeping
it down is the unnecessary musical chairs it makes you go through

[49 more replies]

closed 2024-02-09 by saumyaj3