Ultimaker/Cura #4838: idex collisions

samtaylor3d, 2018-11-19:
**Application Version**
3.6

**Platform**
windows 10

**Printer**
trex2+ from formbot

**Steps to Reproduce**
slicing a file for dual extrusion on trex2+

**Actual Results**
the two extruders collide in the middle

**Expected results**
a beautiful dual coloured print. this is the usual result for cura 3.4.1

**Additional Information**

nallath, 2018-11-20:
Cura doesn't support moving two extruder heads individually. So it's pure luck that it didn't fail on 3.4.1

ChrisTerBeke, 2018-11-20:
You could try BCN's version of Cura. They have an IDEX machine and made specific changes for it compared to Ultimaker's version of Cura (which is this repository).

closed 2018-11-20 by ChrisTerBeke

samtaylor3d, 2018-11-21:
thanks guys. it also worked on cura 15.4.something, and cura 3.3.something
as well. ironically the bcn3d version causes crashes as well.