Ultimaker/Cura #4839: Cura master version doesn't start

joonghochoe, 2018-11-20:
**Application Version**

Cura master version

**Platform**

Windows-10-10.0.17134

**Printer**

**Steps to Reproduce**

Cura master version build -> run

**Actual Results**

Stop at loading plugin...(backend plugin)

**Expected results**

Normal execution

**Additional Information**

Cura 3.6 and last week(11.13 ?) master version are build, execution ok, but current master version will not execute from November 14(?) version. When master version build, there are not any error.

Crash Report
Error:
```
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "Console.py", line 24, in <module>
  File "C:\xxx\cura-build\build\inst\bin\cura_app.py", line 138, in <module>
    app.run()
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 685, in run
    super().run()
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 287, in run
    super().run()
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\Application.py", line 214, in run
    self.startSplashWindowPhase()
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 429, in startSplashWindowPhase
    super().startSplashWindowPhase()
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 179, in startSplashWindowPhase
    self._loadPlugins()
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 680, in _loadPlugins
    raise RuntimeError("Could not load the backend plugin!")
RuntimeError: Could not load the backend plugin!

```
Logs:
```
Thread 0x000027bc (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x00001fb4 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x0000176c (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Thread 0x000006e0 (most recent call first):
  File "threading.py", line 293 in wait
  File "threading.py", line 424 in acquire
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 86 in _nextJob
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 112 in run
  File "threading.py", line 914 in _bootstrap_inner
  File "threading.py", line 882 in _bootstrap

Current thread 0x00000e30 (most recent call first):
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 310 in _logInfoWidget
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 160 in _createDialog
  File "C:\xxx\cura-build\build\inst\lib\python3.5\site-packages\cura\CrashHandler.py", line 81 in __init__
  File "C:\xxx\cura-build\build\inst\bin\cura_app.py", line 116 in exceptHook
```
I deleted the environment folder and tried to reinstall build environment. But the result is the same. Is there any idea?

smartavionics, 2018-11-20:
I discovered this also, to fix it, the cmake in Cura directory needs something like this adding to the command line:

````
-DCURA_SDK_VERSION:STRING=5.0.0
````

smartavionics, 2018-11-20:
You could add that to the CMAKE_ARGS defined in cura-build/projects/Cura.cmake

BTW - I think the layer view is still broken in the master branch - they fucked it up and haven't bothered to fix it.

smartavionics, 2018-11-20:
Actually, I just noticed that the CMAKE_ARGS already has a def for CURA_SDK_VERSION (it has -DCURA_SDK_VERSION=${CURA_SDK_VERSION}) but, obviously, the ${CURA_SDK_VERSION} doesn't have the required value.

Appesteijn, 2018-11-20:
"they fucked it up and haven't bothered to fix it." the internet at its finest...

We are working on a big UI rebuild, a part of layerview in Uranium was accidentally pushed to master. We expect the UI rebuild to be merged to master this sprint.

joonghochoe, 2018-11-20:
@smartavionics Thanks, cura master version build and execution are ok. And layer view doesn't run yet...

Vandrasc, 2018-11-20:
This will be fixed once we merge the new UI rebuild to master, unfortunately, there is nothing we can do now,  lesson learned :)

closed 2018-11-20 by Vandrasc