Ultimaker/Cura #4844: [3.6] Resize tool description is cut

Liger0, 2018-11-21:
**Application Version**
3.6

**Platform**
Win 10 x64 (QHD monitor's resolution)

**Printer**
Cr-10s

**Steps to Reproduce**
Select a model and try the scale tool

**Actual Results**
I can't complretely read what the checkboxes do

**Expected results**
I should be able to understand what they do

**Additional Information**
[image]

Ghostkeeper, 2018-11-21:
Maybe you can provide a better Italian translation for the text "Snap scaling" and "Uniform scaling", that is a bit shorter?

Liger0, 2018-11-21:
> 
> 
> Maybe you can provide a better Italian translation for the text "Snap scaling" and "Uniform scaling", that is a bit shorter?

Hi. I am unsure what the first option actually is for.
About the second one, I'd call it "Scala in proporzione"

closed 2026-09-06 by GregValiant