Ultimaker/Cura #4845: 3.6.0 Printing Bottom When None Exists In Model

brandoncristwright, 2018-11-21:
**Application Version**
Cura 3.6.0

**Platform**
Windows 10

**Printer**
Prussa I3

**Steps to Reproduce**
Load Model, Slice, Print

**Actual Results**
The bottom of the model is supposed to be open but when it prints, it prints a closed bottom.

**Expected results**

**Additional Information**

[image]

smartavionics, 2018-11-21:
Please attach either the zipped project file or the model to this issue, thanks.

diegopradogesto, 2018-11-21:
I'm guessing that the problem is with the mesh itself. If you share the STL we can take a look. You can also check in the X-Ray view if the model has open parts, or use the MeshTools plugins to see if the model is correct.

Sheraphin, 2018-12-16:
I have the same issue. Cura 3.6 is making a bottom where there should be none.

smartavionics, 2018-12-16:
I don't get a layer at the bottom but it does put a weird skin layer in. Model doesn't appear to have any faults so I can't imagine what the problem is here.

[image]

closed 2020-06-20 by no-response[bot]