Ultimaker/Cura #4846: [3.6] model doesn't directly go to the bed when scaling it

Liger0, 2018-11-21:
**Application Version**
3.6

**Platform**
Win 10 x64

**Printer**
Cr-10S

**Steps to Reproduce**
Use a big model. Scale it down.

**Actual Results**
The model is suspended on the air so I have to use the "flat on the bed" option (or how it is called).

**Expected results**
After resizing it should lower to touch the plate directly (If I am not wrong it did it on the previous cura versions)

smartavionics, 2018-11-21:
Or you can just use the move tool and enter 0 into the z field.

diegopradogesto, 2018-11-21:
In preferences, do you have the "Automatically drop models to the build plate" enabled?

It can also be the case that when you scale down the model, it stays outside the buildplate. In this case even with the option enabled, the model will not drop to the buildplate. But this was like this also in previous versions.

Liger0, 2018-11-21:
> In preferences, do you have the "Automatically drop models to the build plate" enabled?
>
> It can also be the case that when you scale down the model, it stays outside the buildplate. In this case [...]

Yes, I am referring to models bigger than the printer. Even if they are touching the bed, when scaling them down they will be suspended in air even with the drop model option enabled.

Liger0, 2018-11-21:
> 
> 
> Or you can just use the move tool and enter 0 into the z field.

The problem is if the models are very big and you scale them down a lot, I have an hard time finding them in the far air as they are too little even to see them and I need to span up for a good amount of time.

fieldOfView, 2018-11-21:
The underlying problem is that objects that are outside the build volume don't automatically drop to the buildplate level. Even if you scale them down, they won't drop down untill you put them in the build volume first.

closed 2020-06-20 by no-response[bot]