Ultimaker/Cura #4847: Layer view slider middle handle broken

BagelOrb, 2018-11-21:
**Application Version**
master @ 24 october
Cura 3.6.0 as well
I've noticed this behavior a couple of months back, but I hoped it would be fixed automatically

**Platform**
Linux mint

**Printer**
UM3

**Steps to Reproduce**
prepare gcode
go to layer view
drag upper layer limit downward a bit
drag lower layer limit upward a bit
click on the middle of the layer range
click on the layer number text box
hit the up or down arrow keys

**Actual Results**
only the max ;ayer of the range changes

**Expected results**
The whole range is transposed,
i.e. both the min and max layer are changed

**Additional Information**
I really like going through my layers while viewing just a couple of them

Viewing all layers at the same time can clutter the UI a bit and slows down my PC

Arrow keys are more accurate than dragging the layer range with the mouse
I sometimes need to go through my layers one by one to be sure the gcode is ok

fieldOfView, 2018-11-21:
Thanks, I meant to report that but I forgot.

rburema, 2018-11-21:
Affects the current UI rework branch as well. I've made a sticky on the board, it'll get discussed tomorrow morning.

rburema, 2018-11-22:
The conclusion was that the behaviour you're suggesting is indeed more logical, but we wont fix this at this time. Feel free to make a PR though, of course.

fieldOfView, 2018-11-22:
It is the behavior that once was. I know, because I implemented it. I would say "You break it, you fix it."

BagelOrb, 2018-11-22:
I agree.

I wasn't asking for a new feature; I was submitting a bug report.